franckbour/Plugin.NFC #1: Add support nfc tag id (serial number)

dm-CaT, 2019-08-08:
### Description
There is no possibility to get nfc tag id (serial number). Sometimes the only id is needed.
### Steps to Reproduce
1. Subscribe to OnMessageReceived event
2. Use NFC card with no data
3. Event is handled

### Expected Behavior
There is an information about NFC card id (serial number) in ITagInfo event argument
### Actual Behavior
ITagInfo event argument is null

### Additional information
Here is some code example how to change Android implementation (I'm not skilled enough in iOS yet).

```
//src/Plugin.NFC/Shared/ITagInfo.shared.cs
// TagInfo class should be changed too
	public interface ITagInfo
	{
		bool IsWritable { get; set; }
		bool IsEmpty { get; }
		NFCNdefRecord[] Records { get; set; }

		// new property
		string SerialNumber { get; set; }
	}
```
	
```
//src/Plugin.NFC/Android/NFC.android.cs

		ITagInfo GetTagInfo(Tag tag, NdefMessage ndefMessage = null)
		{
			if (tag == null)
				return null;
			
			// create nTag and get an id
			var nTag = new TagInfo();
			var nTagIdBytes = tag.GetId();
			var nTagId = nTagIdBytes == null ? 
						string.Empty : 
						string.Concat(nTagIdBytes.Select(b => b.ToString("X2")));
			nTag.SerialNumber = nTagId;

			var ndef = Ndef.Get(tag);

			if (ndef == null)
  				return nTag;
//				return null;

			if (ndefMessage == null)
				ndefMessage = ndef.CachedNdefMessage;

			nTag.IsWritable = ndef.IsWritable;
/*
			var nTag = new TagInfo()
			{
				IsWritable = ndef.IsWritable
			};
*/
			if (ndefMessage != null)
			{
				var records = ndefMessage.GetRecords();
				nTag.Records = GetRecords(records);
			}

			return nTag;
		}

```

franckbour, 2019-08-12:
Ok, I will add this on Android or you can make a PR if you want.
For iOS, Apple/Core NFC does not currently allow it. It seems it will be possible in the next release of iOS (iOS 13).

dm-CaT, 2019-08-19:
If smart-card doesn't contain any records the TagInfo is null.
The TagInfo with serial is expected.

dm-CaT, 2019-08-19:
I cannot create PR because in VS2019 preview 2.0 "MSBuild.Sdk.Extras" is broken.
[https://twitter.com/madrvojt/status/1161666121239617537](https://twitter.com/madrvojt/status/1161666121239617537)

franckbour, 2019-08-19:
Ok, I reopen the issue !

franckbour, 2019-08-22:
I didn't have any problem to get tag identifier even if my nfc tag is empty (no records). 
What kind of nfc tags do you use?

PinoPinillo, 2020-01-08:
Hi, I have problems to read the UUID with iOS 13. It detects the tag, but the problem is that the identifier and the serial number are always null. This doesn't happen in Android.

I'm using the last version of the plugin and I have followed the README instructions, including all the needed code in info.plist and entitlements.plist.

Thanks in advance!

franckbour, 2020-01-10:
Are you using the lastest version from CI feed (0.1.13)? If so, what type of tag do you use?

PinoPinillo, 2020-01-10:
Ups, I'm using the latest version of NuGet, 0.1.11, I'll try with 0.1.13. Anyway, we are using MiFare tags.

franckbour, 2020-01-10:
FYI, another release is available in the CI feed (0.1.14) with some minor changes. 
Feel free to open another issue if you have any problem.

closed 2020-01-10 by franckbour

PinoPinillo, 2020-01-10:
It worked! Thanks for your help!