franckbour/Plugin.NFC #14: OnMessageReceived duplicated

diogopineda, 2020-01-17:
### Description
When I read a nfc tag I get the result, the second time the event OnMessageReceived is called 2 times, 3 times on third and goes on.
Implemented according to Code description
### Steps to Reproduce

1. Read a tag, OnMessageReceived is called
2. Read another Tag, OnMessageReceived is called 2 times.

### Expected Behavior
OnMessageReceived should be called only once by read.
### Actual Behavior
OnMessageReceived is called multiple times.
### Basic Information

View:
```
public NFCSearch (NFCSearchViewModel vm = null)
		{
			InitializeComponent ();
            if (vm != null)
                this.BindingContext = vm;
            else
                this.BindingContext = new NFCSearchViewModel(Navigation);
            // Event raised when a ndef message is received.
            CrossNFC.Current.OnMessageReceived += ((NFCSearchViewModel)this.BindingContext).HandleNewTag;
            CrossNFC.Current.StartListening();

            CrossNFC.Current.OniOSReadingSessionCancelled += Current_OniOSReadingSessionCancelled;
            if (!CrossNFC.Current.IsAvailable)
            {
                App.Current.MainPage.DisplayAlert("Device not supported", "This device does not support NFC!", "Close");
            }
            if (!CrossNFC.Current.IsEnabled)
            {
                App.Current.MainPage.DisplayAlert("NFC disabled", "Please enable NFC on your device config!", "Close");
            }
        }
```
ViewModel:
```
public void HandleNewTag(ITagInfo tagInfo)
        {
            if (tagInfo.Records != null && !string.IsNullOrEmpty(tagInfo.Records[0].Message))
            {
                try
                {
                    string result = Encoding.GetEncoding("UTF-8").GetString(tagInfo.Records[0].Payload);
                    CrossNFC.Current.StopListening();
                    
                    ResultScan = result.Split('/').Last();
                    OpenEqu(ResultScan);
                }
                catch (Exception ex)
                {
                    throw ex;
                }
            }
        }
```

- Version with issue: 0.1.11
- Last known good version: first test
- IDE: VS Pro 2019
- Platform Target Frameworks:
  - iOS:  13
  - Android: 9
- Nuget Packages: 0.1.11
- Affected Devices: Tested on iPhone 7 and Asus Zenfone 5z

franckbour, 2020-02-18:
Do you unsubscribe to the `OnMessageReceived` event anywhere?

diogopineda, 2020-02-27:
@franckbour it worked. Thanks a lot!

closed 2020-02-27 by diogopineda

nabilakhlaque, 2020-05-07:
how do you mean by "unsubscribe" to OnMessageReceived? could you give me an example, please?

franckbour, 2020-05-16:
Check out the `UnsubscribeEvents()` method in `MainPage.xaml.cs` of the sample.
`CrossNFC.Current.OnMessageReceived -= Current_OnMessageReceived` is called to unsubscribe to this event.