franckbour/Plugin.NFC #20: Writing same tag just read

obelixA, 2020-02-20:
Hi

I can read tags one by one without any problems. 
Now I need to write the changed record on the tag at once.
Scenario: one comes with a card (tag) and holds it on the phone. The tag is read, data is checked. card is ok --> write current date (within the data) back to the card.
BUT the card cannot be moved around. this must be done in one step

Can someone guide me, how this is done? I can read, but when I try to write instantly, the card (the data on the card) is unreadable...

thanks in advane
Achim

closed 2020-04-23 by franckbour

obelixA, 2020-04-23:
closing w/o solution? slightly uncool...
there is actually no way to handle that

franckbour, 2020-04-24:
As I already answered you in our last email conversation, this is a very specific use-case that i will not implement for the moment. But you can fork the repo, do what you need and eventually submit a pull request  when done!

obelixA, 2020-04-24:
OK. Thanks for your answer.

Best
Achim

=======================================
acs - it-systems UG Haftungsbeschränkt
Lindenhöhe 32
94127 Neuburg am Inn
---------------------------------------
web: www.acs-it-systems.de
mailto: a.bohmann@acs-it-systems.de
---------------------------------------
Tel: +49 8502 9159905
Fax: +49 8502 9159906
Mobil: +49 171 8884139
---------------------------------------
UStID: DE242236887
HRB 8036
=======================================