franckbour/Plugin.NFC #22: Tag Is Missing Exception

lsoto40, 2020-02-25:
### Description
I am getting an exception when trying to write a tag

What is weird is that the system actually writes the tag after throwing the exception, also not happening on the DEV phone.

The exception is that the 'Tag is missing'

Here is the stack trace I am getting:
Plugin.NFC
NFCImplementation.WriteOrClearMessage (Plugin.NFC.ITagInfo tagInfo, System.Boolean clearMessage, System.Boolean makeReadOnly)
Plugin.NFC
NFCImplementation.PublishMessage (Plugin.NFC.ITagInfo tagInfo, System.Boolean makeReadOnly)

obelixA, 2020-02-28:
Due to not getting ANY answers (seems development has stuck) I need to ask you:
can you read and write the tag without pulling it away between reading and writing?

lsoto40, 2020-02-28:
I am not sure of that, since I don't go back and read it. However I do get the message the writing happened successfully. Not sure how this is happening.

I believe the writing is successful, but the exception is also generated.

franckbour, 2020-04-23:
Do you have a reproduction sample or can you provide more information please ?