franckbour/Plugin.NFC #23: Add support for Mifare NFC tags

jerryvanechelpoel, 2020-02-27:
### Description

The TagInfo does not contain data when scanning a MifareClassic NFC.

### Steps to Reproduce

Scan a MifareClassic NFC by using the OnMessageReceived event

### Expected Behavior

The Records property should contain data.

### Actual Behavior

The Records property is null

### Basic Information

- Version with issue: 0.1.14
- Last known good version: Unknown
- IDE: Visual Studio 2019
- Platform Target Frameworks: .NET Standard 2.0
  - iOS: n/a
  - Android: 9 
  - UWP:  n/a 
- Nuget Packages: 
- Affected Devices: Samsuns S9, OnePlus 3T, OnePlus 6T

### Screenshots
[image]

### Reproduction Link
n/a

zizusoft, 2020-04-04:
Any idea is Mifare Classic tag identifier (the CSN) is being returned (even if the payload record isn't) ?

jerryvanechelpoel, 2020-04-06:
I believe it did. What I noticed is that tag is detected, but the payload is not read out. I tried _NFC Tools_ to make sure that the tag contains data, and it dit, so it must be clearly something in this library.

franckbour, 2020-04-23:
I have two MifareClassic 1k tags and its works fine.

The available technologies of your tag should be `NfcA, MifareClassic, Ndef`.
If you have `NdefFormatable` instead of `Ndef`, your need to format the tag before using this plugin.

[image]

jerryvanechelpoel, 2020-04-23:
Well that's exactly what I said: NFC Tools detects the data, but Plugin.NFC just isn't able to read the data from the tag and show it to the consuming code.

franckbour, 2020-04-23:
Then i need more information. Like I just said, i tested with two differents tags and it works fine, data and serial number are read.
What technologies are used in your tag and what kind of data is stored ?

twsl, 2020-05-16:
In order to support mifare classic on android, check [here](https://developer.android.com/reference/android/nfc/tech/MifareClassic), for mifare desfire you need to read from [isodep](https://developer.android.com/reference/android/nfc/tech/IsoDep)

dalton5, 2021-01-02:
Hi,

I had the same issue. And the problem came from the format method line 248

In the NFC.android.cs file,

Add an else and this code. It worked perfectly for me. And I was not able to do it with the original code.

`NdefFormatable ndefFormatable = NdefFormatable.Get(_currentTag);
					if (ndefFormatable != null)
					{
						// initialize tag with new NDEF message
						try
						{
							ndefFormatable.Connect();
							ndefFormatable.Format(null);
						}
						finally
						{
							try
							{
								ndefFormatable.Close();
							}
							catch (Exception e)
							{
								isnull = true;
							}
						}
					}`