franckbour/Plugin.NFC #28: How do I cange the default text?

nabilakhlaque, 2020-05-05:
Hello, I want to know how I change the default NFCDialogAlertMessage?

franckbour, 2020-05-16:
Sorry, This is not yet implemented.

saamerm, 2020-05-16:
@franckbour this is how NFCDialogAlertMessage is consumed in the `NFC.iOS.cs` currently:
```
/// <summary>
/// Starts tags detection
/// </summary>
public void StartListening()
{
	_customInvalidation = false;
	_isWriting = false;
	_isFormatting = false;

	NfcSession = new NFCTagReaderSession(NFCPollingOption.Iso14443 | NFCPollingOption.Iso15693, this, DispatchQueue.CurrentQueue)
	{
		AlertMessage = UIMessages.NFCDialogAlertMessage
	};
	NfcSession?.BeginSession();
}
```
How would you like the change to be made?

### First way we can solve: 
We can either update the interface/API in INFC.shared.cs to this:
`void StartListening(string listeningMessage = "");` 
and then change the AlertMessage line in the `StartListening()` from NFC.iOS.cs shown above, to this:
`AlertMessage = string.IsNullOrEmpty(listeningMessage)? listeningMessage : UIMessages.NFCDialogAlertMessage`

### Second way we can solve: 
Convert NFCDialogAlertMessage in UIMessages to be a static variable that can be edited:
```
public static string NFCDialogAlertMessage = "Please hold your phone near an NFC tag";
```
and create a OverrideiOSDialogAlertMessage() in INFC.shared.cs 
```
void OverrideiOSDialogAlertMessage(string dialogAlertMessage);
```
 that overrides the value of NFCDialogAlertMessage
```
public void OverrideiOSDialogAlertMessage(string dialogAlertMessage) =>
			UIMessages.NFCDialogAlertMessage = dialogAlertMessage;
```

franckbour, 2020-05-17:
I think the second way is better because there are other messages that users might want to change.

We could add a new configuration object like below and pass it to the `CrossNFC.Current` instance via a method like `CrossNFC.Current.SetConfiguration(cfg)`

```csharp
public class NfcConfiguration
{
    public UserDefinedMessages Messages { get; set; }
}

public class UserDefinedMessages
{
    public string NFCDialogAlertMessage { get; set; } = "Please hold your phone near a NFC tag";
    public string NFCErrorRead { get; set; } = "Read error. Please try again";
    public string NFCErrorEmptyTag { get; set; } = "Tag is empty";
    public string NFCErrorReadOnlyTag { get; set; } = "Tag is not writable";
    public string NFCErrorCapacityTag { get; set; } = "Tag's capacity is too low";
    public string NFCErrorMissingTag { get; set; } = "Tag is missing";
    public string NFCErrorMissingTagInfo { get; set; } = "No Tag Informations: nothing to write";
    public string NFCErrorNotSupportedTag { get; set; } = "Tag is not supported";
    public string NFCErrorNotCompliantTag { get; set; } = "Tag is not NDEF compliant";
    public string NFCErrorWrite { get; set; } = "Nothing to write";
    public string NFCSuccessRead { get; set; } = "Tag Read Success";
    public string NFCSuccessWrite { get; set; } = "Tag Write Success";
    public string NFCSuccessClear { get; set; } = "Tag Clear Success";
}
```
What do you think?

saamerm, 2020-05-17:
I agree! This looks solid

closed 2020-06-20 by franckbour