franckbour/Plugin.NFC #33: You must disable foreground dispatching while your activity is still resumed

kberawala, 2020-05-20:
### Description
Had anyone seen this ?
"You must disable foreground dispatching while your activity is still resumed"

On Android - I need nfc scanning to listen for the tag only when button is pressed.
iOS - works ok for below code.

### Steps to Reproduce
1.  Android app, when listening for tag one after another.
2. below is the code from view model

```
	private async Task ScanAPointAsync()
        {
            IsBusy = true;
            try
            {
                string[] options = new string[] { "Scan Barcode" };
                if (CrossNFC.IsSupported && CrossNFC.Current.IsAvailable)
                {
                    options = new string[] { "Scan Barcode", "Read a Tag" };
                }

                var result = await page.DisplayActionSheet("Select", "Cancel", null, options);
                if (result == "Read a Tag")
                {
                    if (CrossNFC.Current.IsEnabled)
                    {
                        CrossNFC.Current.OnMessageReceived += Current_OnMessageReceived;
                        CrossNFC.Current.StartListening();
                    }
                    else
                    {
                        await page.DisplayAlert("Enable NFC", "Please enable NFC on your phone.", "OK");
                    }
                }
            }
            catch (Exception)
            {
                await page.DisplayAlert("Error", "Something went wrong, Please try again.", "OK");
            }
            finally
            {
                IsBusy = false;
            }
        }

	private async void Current_OnMessageReceived(ITagInfo tagInfo)
        {
            if (tagInfo != null)
            {
		string serialNumber = tagInfo.SerialNumber;
		
		// do something more here

                CrossNFC.Current.OnMessageReceived -= Current_OnMessageReceived;
                CrossNFC.Current.StopListening();
            }
        }
```

### Expected Behavior
 shouldnt crash

### Actual Behavior
App Crashes with
"You must disable foreground dispatching while your activity is still resumed"

### Basic Information

- Version with issue: 0.1.16
- Last known good version: NA
- IDE: Visual Studio
- Platform Target Frameworks:
   - Android: 10.0

franckbour, 2020-05-21:
You probably need to unsubscribe and stop listening in the main thread like this:
```csharp
Device.BeginInvokeOnMainThread(() =>
{
    CrossNFC.Current.OnMessageReceived -= Current_OnMessageReceived;
    CrossNFC.Current.StopListening();
});
```

kberawala, 2020-05-21:
> You probably need to unsubscribe and stop listening in the main thread like this:
> 
> ```cs
> Device.BeginInvokeOnMainThread(() =>
> {
>     CrossNFC.Current.OnMessageReceived -= Current_OnMessageReceived;
>     CrossNFC.Current.StopListening();
> });
> ```

It worked. Thanks

closed 2020-05-21 by kberawala

pvanroos, 2021-05-31:
I added the above code at the end of the 'Current_OnMessageReceived' event handler and I'm still getting the error:  You must disable foreground dispatching while your activity is still resumed

kberawala, 2021-05-31:
> I added the above code at the end of the 'Current_OnMessageReceived' event handler and I'm still getting the error: You must disable foreground dispatching while your activity is still resumed

check if your page is  => var page = new navigationpage(new scanpage());
and not var page = new scanpage(); (this didn't work for me)

saamerm, 2022-05-08:
This solution worked for me too. @pvanroos were you able to figure out your issue?

LuigiCroni, 2023-09-05:
> check if your page is => var page = new navigationpage(new scanpage()); and not var page = new scanpage(); (this didn't work for me)

scan worked on my homepage but not certain pages, however it worked on a page which used 
"await Navigation.PushAsync(new pageName());" but now that i've changed my way to navigate , it works on every pages.

do we know why this way works and not the other?

nanorie, 2024-01-14:
instead of stop listening directly in Current_OnMessageReceived, i call it another task not waited and it's work on android and Maui
`
        private async void Current_OnMessageReceived(ITagInfo tagInfo)
        {
                // to stuff
                await Task.Run(() => StopListening()).ConfigureAwait(false);
        }
        private async Task StopListening()
        {
            try
            {
                MainThread.BeginInvokeOnMainThread(() =>
                {
                    CrossNFC.Current.StopListening();
                    CrossNFC.Current.OnMessageReceived -= Current_OnMessageReceived;
                });
            }
            catch (Exception ex)
            {
            }
        }
`
All this code is in a service and works for me (but I don't understand the underlying mechanism that makes it work)
Hope it can help someone