franckbour/Plugin.NFC #34: Issue with NFC on Android

nabilakhlaque, 2020-05-21:
### Description
I cant seem to make it work, I've followed the instructions, but nothing happens when CrossNFC.Current.StartListening(); is called. On iOS if work like a charm. I have added these line inside <manifest> -tag:

<uses-permission android:name="android.permission.NFC" />
<uses-feature android:name="android.hardware.nfc" android:required="false" />

I have also init it in MainActivity.cs in onCreate()

My manifest file:
`<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="41" android:versionName="1.41" package="se.edgeteq.loyiz" android:installLocation="auto">
	<uses-sdk android:minSdkVersion="22" android:targetSdkVersion="29" />
	<uses-permission android:name="android.permission.VIBRATE" />
	<uses-permission android:name="android.permission.INTERNET" />
	<uses-permission android:name="android.permission.CAMERA" />
	<uses-permission android:name="android.permission.FLASHLIGHT" />
	<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
	<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_LOCATION_EXTRA_COMMANDS" />
	<uses-permission android:name="android.permission.ACCESS_MOCK_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
	<uses-permission android:name="android.permission.GET_ACCOUNTS" />
	
	<uses-permission android:name="android.permission.NFC" />
	<uses-feature android:name="android.hardware.nfc" android:required="false" />
	
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	
	<permission android:name="se.edgeteq.loyiz.permission.C2D_MESSAGE" android:protectionLevel="signature" />
	<application android:label="Mingo" android:icon="@mipmap/ic_launcher" android:usesCleartextTraffic="true">
		<uses-library android:name="org.apache.http.legacy" android:required="false" />
		<meta-data android:name="com.google.android.geo.API_KEY" android:value="[HIDDEN]" />
		<receiver android:name="com.google.firebase.iid.FirebaseInstanceIdInternalReceiver" android:exported="false" />
		<receiver android:name="com.google.firebase.iid.FirebaseInstanceIdReceiver" android:exported="true" android:permission="com.google.android.c2dm.permission.SEND">
			<intent-filter>
				<action android:name="com.google.android.c2dm.intent.RECEIVE" />
				<action android:name="com.google.android.c2dm.intent.REGISTRATION" />
				<category android:name="${applicationId}" />
			</intent-filter>
		</receiver>
		<provider android:name="android.support.v4.content.FileProvider" android:authorities="${applicationId}.fileprovider" android:exported="false" android:grantUriPermissions="true">
			<meta-data android:name="android.support.FILE_PROVIDER_PATHS" android:resource="@xml/file_paths"></meta-data>
		</provider>
		<meta-data android:name="com.facebook.sdk.ApplicationId" android:value="@string/facebook_app_id" />
		<activity android:name="com.facebook.FacebookActivity" android:configChanges="keyboard|keyboardHidden|screenLayout|screenSize|orientation" android:label="@string/facebook_app_name" />
		<activity android:name="com.facebook.CustomTabActivity" android:exported="true">
			<intent-filter>
				<action android:name="android.intent.action.VIEW" />
				<category android:name="android.intent.category.DEFAULT" />
				<category android:name="android.intent.category.BROWSABLE" />
				<data android:scheme="@string/fb_login_protocol_scheme" />
			</intent-filter>
		</activity>
	</application>
</manifest>`

### Steps to Reproduce

1. 
2. 
3. 

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue: 0.1.15
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

kberawala, 2020-05-21:
just checking have you done this ?

[image]

franckbour, 2020-05-21:
Do you have a reproduction sample?

nabilakhlaque, 2020-05-21:
> 
> 
> just checking have you done this ?
> 
> [image]

yes I have.

nabilakhlaque, 2020-05-21:
> 
> 
> Do you have a reproduction sample?

Not right now, but I can create a new sameple project and share it with you.

nabilakhlaque, 2020-05-21:
Here I have created a new project from scratch... I hope its just something small that Im missing... 
[https://send.firefox.com/download/6f92abcb04b3bc9d/#EM3BlwhlsgdY_sY0JO0wIA](url)

franckbour, 2020-05-21:
In iOS, it shows a dialog when you're listening for a tag. 

In Android, you don't have any visual indication, it just starts the listener in the background so when you'll present a tag, it will be detected and read.

Your sample is not working. I got the same error than issue #33 when I present a tag.

In order to make it work, you can remove `OnMessageReceived` subscription from `MainPage()` constructor and replace this code:

```csharp
void Current_OnMessageReceived(ITagInfo tagInfo)
{
    CrossNFC.Current.StopListening();

    try
    {
        string accountKey = tagInfo.Records[0].Message;
        Debug.WriteLine(accountKey);
    }
    catch (Exception ex)
    {

    }
}

private void Button_Clicked(object sender, EventArgs e)
{
    CrossNFC.Current.StartListening();
}
 ```

by this one:
```csharp
void Current_OnMessageReceived(ITagInfo tagInfo)
{
    try
    {
        string accountKey = tagInfo.Records[0].Message;
        Debug.WriteLine(accountKey);
    }
    catch (Exception ex)
    {
        Debug.WriteLine(ex.StackTrace);
    }
    finally
    {
        Device.BeginInvokeOnMainThread(() =>
        {
            // Unsubscribe from OnMessageReceived!
            CrossNFC.Current.OnMessageReceived -= Current_OnMessageReceived;

            CrossNFC.Current.StopListening();
        });
    }
}

void Button_Clicked(object sender, EventArgs e)
{
    // Subscribe to OnMessageReceived here!
    CrossNFC.Current.OnMessageReceived += Current_OnMessageReceived;

    CrossNFC.Current.StartListening();
}
```
Let me know if it solves your problem.

closed 2020-07-11 by franckbour