franckbour/Plugin.NFC #37: Send class from phone to phone

Dimtrivh, 2020-06-12:
### Description

A Question

Is there no way to send data from 1 phone to another with nfc, android beam.

I would like to exchange data(maybe a json string) when 2 phones come close to each other

Thx for the advice

### Steps to Reproduce

1. 
2. 
3. 

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue:
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

saamerm, 2020-07-12:
You are looking at "HCE" or Host-based Card Emulation. 

It seems to be [possible but difficult](https://stackoverflow.com/questions/24916259/use-another-android-device-as-nfc-tag) to do on Android

And you certainly can't do it on iOS devices yet, as you can see from the [Apple forums](https://developer.apple.com/forums/thread/128931).

saamerm, 2020-11-19:
@Dimtrivh

closed 2021-04-19 by franckbour