franckbour/Plugin.NFC #40: [iOS] NFC Tag not reading on iOS unless I have a message on the tag

GameJaxx, 2020-06-30:
### Description

Basically the title. I can't read a NFC on iOS unless it has a message on it. For the purposes of my app, I'm only really wanting to get the "SerialNumber" from the scan and thus, leave the tag empty.

I've got the code working as intended on Android with both a empty tag and tag with a "test" message on it working.

On iOS, if I scan the tag with the "test" message it works but I get "stack error" or "read error please try again" on an empty tag. I'm using OnMessageReceived which I thought might be the issue due to its naming but as I said, on Android with an empty tag, I get the intended output.

### Expected Behaviour

Scan the NFC tag and the corresponding OnMessageReceived method would run, giving me the tag's SerialNumber (does so in Android)

### Actual Behaviour

None of the events (OnMessageReceived, even tried "OnTagDiscovered") fire and I'm left with either "stack error" or "read error please try again" errors in the GUI.

### Basic Information

- Version with issue: 0.1.16
- Last known good version: N/A
- IDE:
- Platform Target Frameworks:
  - iOS:  13.5.1
  - Android: 7.0 
Nuget Packages:
- Affected Devices:
Tested on a HUAWEI P10 Plus
Tested on an iphone 7

### Screenshots
https://i.imgur.com/fl0mYva.png - Tag info in case that's needed

Could be missing something obvious so apologies if that's the case and thanks for any help you's can provide.

Lydecker, 2020-07-03:
I get exactly the same error.

I have tags that I want to get the CSN / UDID off of (but there is no data written to the tag).

On other IOS apps - I can read that data off too - with this plugin - I just get a Stack Error or a Read Error.....

saamerm, 2020-07-12:
@Lydecker and @GameJaxx could you guys try to clone this project and try it with the sample herein? And then place an exception catchpoint and let us know where the exception is caught.

closed 2020-07-25 by franckbour