franckbour/Plugin.NFC #41: Please add ISO15693 tags

iDiabDev, 2020-06-30:
### Description
Please add ISO15603 tags support - to iOS
### Steps to Reproduce

1. 
2. 
3. 

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue:
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

saamerm, 2020-07-12:
What are the changes needed to be made in order to support ISO15603 tags?

sujeetulli, 2020-08-10:
Any update on this ticket?

saamerm, 2020-08-10:
@sujeetulli do you know what are the changes needed to be made in order to support ISO15603 tags?

saamerm, 2020-08-10:
@sujeetulli as you can see here https://github.com/franckbour/Plugin.NFC/blob/3a94b5fc8ad64847b296d33ec1abf3a1432f6981/src/Plugin.NFC/iOS/NFC.iOS.cs#L102 , it should work. Could you please share what error message you are getting?
Alternatively, you can work or download this project, and try to see how you can fix it. And then you can submit a pull request with the fix

sujeetulli, 2020-08-11:
> @sujeetulli as you can see here
> 
> https://github.com/franckbour/Plugin.NFC/blob/3a94b5fc8ad64847b296d33ec1abf3a1432f6981/src/Plugin.NFC/iOS/NFC.iOS.cs#L102
> 
> , it should work. Could you please share what error message you are getting?
> Alternatively, you can fork or download this project, and add it to your project and try to see how you can fix it. And then you can submit a pull request with the fix

While adding the below code getting typecast issue.

INFCNdefTag GetNdefTag(INFCTag tag)
		{
			if (tag == null || !tag.Available)
				return null;

			INFCNdefTag ndef;
			if (tag.GetNFCMiFareTag() != null)
				ndef = tag.GetNFCMiFareTag();
			else if (tag.GetNFCIso7816Tag() != null)
				ndef = tag.GetNFCIso7816Tag();
			else if (tag.GetNFCFeliCaTag() != null)
				ndef = tag.GetNFCFeliCaTag();
			**else if (tag.GetNFCIso15693Tag() != null)
				ndef = tag.GetNFCIso15693Tag();**
			else
				ndef = null;

			return ndef;
		}

Cannot implicitly convert type 'CoreNFC.INFCIso15693Tag' to 'CoreNFC.INFCNdefTag'. An explicit conversion exists (are you missing a cast?)

sujeetulli, 2020-08-12:
@saamerm @franckbour Do you have a solution for this?
Otherwise, I need to look into alternative ways.

saamerm, 2020-08-12:
@sujeetulli unfortunately I have no NFCIso15693Tag device to replicate the issue. 
And why don't you just use the `is` keyworkd instead, like this:
`else if (tag is INFCIso15693Tag iso15693Tag)`

sujeetulli, 2020-08-13:
@saamerm  I'm already done with above changes, 
But my only point is, How can we read 'message'
if (result is INFCIso15693Tag iso15693Tag)

Example: If i use the below NFCNdef then i'm able to read the data.
public class NFCImplementation_Before_iOS13 : **NFCNdefReaderSessionDelegate**, INFC
/// <summary>
		/// Event raised when NDEF messages are detected
		/// </summary>
		/// <param name="session">iOS <see cref="NFCNdefReaderSession"/></param>
		/// <param name="messages">Array of iOS <see cref="NFCNdefMessage"/></param>
		public override void DidDetect(NFCNdefReaderSession session, **NFCNdefMessage[] messages**)
		{
			OnTagConnected?.Invoke(null, EventArgs.Empty);

			if (messages != null && messages.Length > 0)
			{
				var first = messages[0];
				var tagInfo = new TagInfo
				{
					IsWritable = false,
					Records = GetRecords(**first.Records**)
				};
				OnMessageReceived?.Invoke(tagInfo);
			}

			OnTagDisconnected?.Invoke(null, EventArgs.Empty);
		}

So here, What is way to get the records from tag.

f-richter, 2020-11-19:
> > @sujeetulli as you can see here
> > https://github.com/franckbour/Plugin.NFC/blob/3a94b5fc8ad64847b296d33ec1abf3a1432f6981/src/Plugin.NFC/iOS/NFC.iOS.cs#L102
> > 
> > , it should work. Could you please share what error message you are getting?
> > Alternatively, you can fork or download this project, and add it to your project and try to see how you can fix it. And then you can submit a pull request with the fix
> 
> @franckbour @saamerm
> 
> While adding the below code getting typecast issue.
> 
> INFCNdefTag GetNdefTag(INFCTag tag)
> {
> if (tag == null || !tag.Available)
> return null;
> 
> ```
> 		INFCNdefTag ndef;
> 		if (tag.GetNFCMiFareTag() != null)
> 			ndef = tag.GetNFCMiFareTag();
> 		else if (tag.GetNFCIso7816Tag() != null)
> 			ndef = tag.GetNFCIso7816Tag();
> 		else if (tag.GetNFCFeliCaTag() != null)
> 			ndef = tag.GetNFCFeliCaTag();
> 		**else if (tag.GetNFCIso15693Tag() != null)
> 			ndef = tag.GetNFCIso15693Tag();**
> 		else
> 			ndef = null;
> 
> 		return ndef;
> 	}
> ```
> 
> Cannot implicitly convert type 'CoreNFC.INFCIso15693Tag' to 'CoreNFC.INFCNdefTag'. An explicit conversion exists (are you missing a cast?)

This suggested change actually works for me and now I am able to read and write ISO 15693 tags.

saamerm, 2020-11-19:
@f-richter That's awesome! So if you add these two lines you are able to read ISO 15693 tags?
```
		else if (tag.GetNFCIso15693Tag() != null)
			ndef = tag.GetNFCIso15693Tag();
```
No other changes? Do you want to create a PR?

f-richter, 2020-11-19:
Exactly, these two lines. I would like to create a PR, but I don't have the permission.

saamerm, 2020-11-19:
You should be able to, just like I did when I first contributed. Following is the summary, and you can see details [here](https://opensource.com/article/19/7/create-pull-request-github).

* Fork it
* Clone it to your local system
* Make a new branch
* Make your changes
* Push it back to your repo
* Click the Compare & pull request button
* Click Create pull request and voilà!

It's easy for me to do, but it will be good practice for you if you like! Just let me know @f-richter

f-richter, 2020-11-19:
PErfect, thank you @saamerm. PR is created https://github.com/franckbour/Plugin.NFC/pull/61

closed 2020-12-08 by franckbour