franckbour/Plugin.NFC #42: [ BUG ] Java.Lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed

Jakar510, 2020-07-02:
### Description

When scanning an NFC tag on Android 10, I get the following exception:

	Java.Lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
	  at Java.Interop.JniEnvironment+InstanceMethods.CallVoidMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x0006e] in <9324da45a6654f83baffa7c2854d836a>:0 
	  at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeAbstractVoidMethod (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x00014] in <9324da45a6654f83baffa7c2854d836a>:0 
	  at Android.Nfc.NfcAdapter.EnableForegroundDispatch (Android.App.Activity activity, Android.App.PendingIntent intent, Android.Content.IntentFilter[] filters, System.String[][] techLists) [0x0008e] in <eaa205f580954a64824b74a79fa87c62>:0 
	  at Plugin.NFC.NFCImplementation.StartListening () [0x0008f] in <aa6103da0da343a6ad2d3f737fa2443d>:0 
	  at {Private}.Scanner.<StartNFC>b__53_0 () [0x00030] in {Private}.Scanner.cs:305 
	  at Xamarin.Essentials.MainThread.BeginInvokeOnMainThread (System.Action action) [0x00007] in d:\a\1\s\Xamarin.Essentials\MainThread\MainThread.shared.cs:16 
	  at {Private}.Scanner.StartNFC () [0x00001] in {Private}.Scanner.cs:297 
	  at {Private}.Scanner.AfterNFC (System.Threading.CancellationToken token) [0x0010e] in {Private}.Scanner.cs:274 
	  at {Private}.LoadingAsyncTask (System.Func`2[T,TResult] func, System.String title, System.String cancel, Acr.UserDialogs.MaskType mask) [0x000d3] in {Private}:88 
	  --- End of managed Java.Lang.IllegalStateException stack trace ---
	java.lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed
		at android.nfc.NfcAdapter.enableForegroundDispatch(NfcAdapter.java:1787)
		at crc64d79a34a3c547b7ed.MainActivity.n_onNewIntent(Native Method)
		at crc64d79a34a3c547b7ed.MainActivity.onNewIntent(MainActivity.java:55)
		at android.app.Activity.performNewIntent(Activity.java:7971)
		at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1407)
		at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1420)
		at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:3779)
		at android.app.ActivityThread.handleNewIntent(ActivityThread.java:3791)
		at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:53)
		at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
		at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
		at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2229)
		at android.os.Handler.dispatchMessage(Handler.java:107)
		at android.os.Looper.loop(Looper.java:237)
		at android.app.ActivityThread.main(ActivityThread.java:8034)
		at java.lang.reflect.Method.invoke(Native Method)
		at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
		at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)

For a fix, see [StackOverFlow Post](https://stackoverflow.com/questions/32116531/error-unable-to-start-activity-java-lang-illegalstateexception-foreground-disp)

See [Android Documentation](https://developer.android.com/reference/android/nfc/NfcAdapter#enableForegroundDispatch(android.app.Activity,%20android.app.PendingIntent,%20android.content.IntentFilter[],%20java.lang.String[][])) for more details.

### Steps to Reproduce

1. scan tag on Android 10 device
2. boom

### Expected Behavior
Scan the tag. Started getting this exception and now I can't get it to work again.

### Actual Behavior
The exception is thrown.

### Basic Information

- Version with issue: 0.1.16
- Last known good version: Unknown
- IDE: Visual Studio 2019 v16.6.3
- Platform Target Frameworks:
  - iOS: Untested 
  - Android: 10.0  
  - UWP: Untested  
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

Jakar510, 2020-07-06:
[image]
[image]

Jakar510, 2020-07-06:
Added a test page. must wrap in 'Task.Run(() => MainThread.BeginInvokeOnMainThread(() => ...' in order to re-enable scanning.

[TestNFC.zip](https://github.com/franckbour/Plugin.NFC/files/4881138/TestNFC.zip)

saamerm, 2020-07-12:
@Jakar510 did you take a look at this fix mentioned [over here](https://github.com/franckbour/Plugin.NFC/issues/33)?

closed 2020-07-15 by Jakar510