franckbour/Plugin.NFC #44: Issues Reading Mifare Classic 1k? Read this...

zizusoft, 2020-07-16:
If you are having issues reading Mifare 1k Classic cards - the chances are the issue is not with this library, but with Apple's API.

On iOS 11, apple released the ability to READ NCC NDEF data only using the NfcNdefReaderSession API (https://developer.apple.com/documentation/corenfc/nfcndefreadersession)

A Mifare 1k Classic card will scan if there is a valid NDEF record on it. A blank card will not scan.

In iOS 11, it was not possible to obtain the CSN (serial number/identity) from NFC tags/card.

With iOS 13, along came the ability to write NDEF data AND read serial numbers. However, rather then adapting the NfcNdefReaderSession API, Apple created a NEW API called NfcTagReaderSession (https://developer.apple.com/documentation/corenfc/nfctagreadersession) and left the old NfcNdefReaderSession API untouched.

The new NfcTagReaderSession API in iOS 13 no longer supports Mifare Classic 1k cards period. No idea why - but if you look at Apple's Dev Forums multiple people have spotted the same thing.

So even if you have a Mifare Classic 1k card which reads fine with the old iOS 11 NfcNdefReaderSession API, that same card will not even scan with iOS 13's NfcTagReaderSession API.

If you need to read NDEF data off of a Mifare 1k Classic card, then you need to use version 0.1.11 of this library as it was written with the NfcNdefReaderSession API. Any subsequent versions moved over to the NfcTagReaderSession API

Unfortunately, even with iOS 13, it seems there is no way to read the serial number / CSN off of a Mifare Classic 1k card.

franckbour, 2020-07-25:
Thanks @zizusoft

danielmeza, 2021-02-24:
@franckbour this should be included in the readme?

saamerm, 2021-02-24:
@danielmeza yeah that's a great idea. You want to submit a PR to add it in the README?

zizusoft, 2021-02-24:
In an ideal world, (although I appreciate this would be more work) this library would implement both NfcNdefReaderSession and NfcTagReaderSession.

In other (non Xamarin) libraries they have kept the NfcNdefReaderSession available from iOS11s API to support Mifare 1K Classic (as it exists in abundance in the world due to the low cost) and added the NfcTagReaderSession to the side of it rather then replacing it.

The developer can choose whether they want to fire up an instance of NfcNdefReaderSession or NfcTagReaderSession depending on their target audience. Looks like most other libraries (again, non Xamarin) are calling the iOS11 implementation 'Compatibility Mode'.

saamerm, 2021-02-24:
@zizusoft you're welcome to submit a PR so we can support both

danielmeza, 2021-03-01:
@zizusoft @franckbour @saamerm Tanks you! Can this be closed now?

closed 2021-05-28 by franckbour