franckbour/Plugin.NFC #50: IOS 13.6 IPhone 11 Pro Max wont scan IOS 14443-4 NXP MiFare Plus

firstcoder-1, 2020-08-11:
### Description
Hi Trying to use this framework but not working for: IOS 13.6 IPhone 11 Pro Max, wont scan IOS 14443-4 NXP MiFare Plus 
-Works fine on Andriod, 
-It wont scan using "NFCTagReaderSession"
-it will read using "NFCNdefReaderSession", tried another stand along project

Any ideas? trying to use "NFCTagReaderSession", to write to tag. 

### Steps to Reproduce

1. Build/run app on IOS (13.0+)13.6 IPhone 11 Pro Max
2. Click read tag
3. Popup shows "Please hold your phone near a NFC tag"
4. Hover phone on top of tag, but never reads it.

### Expected Behavior
Should read a the tag

### Actual Behavior
Never reads it.

### Basic Information
Using NFC Tools App to scan the same tag says its a = IOS 14443-4 NXP MiFare Plus tag

- Version with issue:
- Last known good version:
- IDE: Visual Studio for Mac
- Platform Target Frameworks: IOS13, IOS13.6
  - iOS:  IOS13, IOS13.6 (builds and runs but wont scan tag)
  - Android: 9, works fine
  - UWP: N/A
- Nuget Packages: no changes exactly as in your solution
- Affected Devices: IPhone 11 Pro Max

### Screenshots

### Reproduction Link

ahmedkasani1988, 2020-08-29:
did you get solution please?

firstcoder-1, 2020-08-29:
The current solution on further testing worked with ISO 14443-3 tags on IOS 13.6 IPhone 11 Pro Max
but for ISO 14443-4 tags, its not working.
The best work around after debugging the problem in Swift code, was to use NFCNdefReaderSession as this is how all the problems in Swift are done (from all apples read & write examples), which works both read and write for 14443-3 & 14443-4. This worked for my current project application which is all I needed it for so I will close issue.

closed 2020-08-29 by firstcoder-1

ahmedkasani1988, 2020-08-30:
My thanks to you sir