franckbour/Plugin.NFC #51: How can I write an NFC tag that an RFID reader understands?

KerosenoDev, 2020-08-20:
### Description
I have the RFID reader RD200-M1-G, which is compatible with: MIFARE 1K S50, 4K S70 4/7-byte UID and MIFARE Ultralight.
I use the sample trying to write an NFC tag that the reader can understand (hex):

[image]

But it doesn't read the code I want, instead random codes come out:

[image]

It's possible? Thank you.

saamerm, 2020-09-07:
Did you try a different mime type or type format? Are you able to use any apps that are already published on the app store?

saamerm, 2020-09-22:
@KerosenoDev ^

saamerm, 2020-11-19:
@franckbour perhaps close this from inactivity?

closed 2020-11-23 by KerosenoDev