franckbour/Plugin.NFC #52: Enable setting LanguageCode for TextRecords

ManniAT, 2020-08-20:
### Description

### Steps to Reproduce

1. Use the sample to write a TextRecord on Android
2. 
3. 

### Expected Behavior
Have a possibility to set the Language code when creating a TextRecord

### Actual Behavior
Uses Locale.Default.ToLanguageTag() (NSLocale.CurrentLocale on IOS) which obtains the current (set on the device) language.

I want to create Tags with specific (different) LanguageCodes on a device.

A simple way would be that you make the methods GetiOSPayload / GetAndroidNdefRecord virtual so it would be easy to enhance the record creation.

saamerm, 2020-09-22:
@ManniAT could you create a draft PR for this work and we ll look at possibly integrating it

ManniAT, 2020-10-07:
@saamerm sorry for the delay - I'll do this the next days.

ManniAT, 2020-10-07:
@saamerm I cloned the repo - changed code, created a branch "NDEFLocale" - but I'm not allowed to push
>>remote: Permission to franckbour/Plugin.NFC.git denied to ManniAT.

saamerm, 2020-10-07:
@ManniAT yup, you have to create a pull request from your own repo. Here’s a resource: https://www.earthdatascience.org/courses/intro-to-earth-data-science/git-github/github-collaboration/how-to-submit-pull-requests-on-github/

ManniAT, 2020-10-07:
Sorry for that :-)

closed 2020-12-01 by franckbour