franckbour/Plugin.NFC #53: Doesn't work on Android

waletoye, 2020-09-08:
### Description
I installed the sample project on two andorid phones (Samsung s10+ and Note 4). It keeps giving the exact same information (as in the attachment) no matter what i try to "write" to either phone. 

### Steps to Reproduce

1. Install on each phone 
2. Attempt to write to either phone 
3. Attempt to read from either phone. 

### Expected Behavior
1. Tags should be written successfully. 
2. Tags should be read successfully with the correct message written on the tag. 

### Actual Behavior
When writing or reading, sometimes i get either:
1. Tag does not support NDEF format. 
2. No tag. 
3. Or the same message as attached

### Basic Information

- Version with issue:
Latest version. 

- Last known good version:
- IDE:
Vs for mac.

- Platform Target Frameworks:
  - iOS:  (not attempted yet) 

  - Android: android 6 and 10
 
- Nuget Packages:
- Affected Devices:
Samsung galaxy s10+
Samsung note 4

### Screenshots
[image]

### Reproduction Link

saamerm, 2020-09-22:
You’re saying that all button perform a Read function in the sample app? @waletoye 
Or is it an issue in the library itself?

saamerm, 2020-11-19:
@waletoye could you please test the latest sample?

saamerm, 2021-04-13:
@waletoye is this still an issue, or would you like to close this?

waletoye, 2021-04-13:
@saamerm I realised Samsung s10+ tag are read-only.
I will close this now.

thanks.

closed 2021-04-13 by waletoye