franckbour/Plugin.NFC #54: Android: Event lost when press home button

franksfleung, 2020-09-14:
### Description
The tag cannot read after press home button

### Steps to Reproduce

1. Open the app
2. Tag card 
3. Press home button
4. Wakeup the app
5. Tag card

### Expected Behavior
Card should be read at step 5.

### Actual Behavior
The app does not has response at step 5.

I used this as reference
https://www.uuu.com.tw/Public/content/article/20/20200427.htm

saamerm, 2020-09-22:
Did you try to press the read tag button, and did it not tag the card again? @franksfleung

franksfleung, 2020-09-23:
Yes it works if press the read tag button.
But I expected it can read the card even I do not click read tag button.

saamerm, 2020-09-23:
Try to get the most recent changes that were just merged @franksfleung

saamerm, 2020-11-19:
@franksfleung is this still an issue?

franksfleung, 2020-11-27:
@saamerm not yet try. Hope can test at coming weeks

franksfleung, 2020-12-30:
@saamerm Just download the latest version, but still fail

franckbour, 2021-02-09:
@franksfleung do you reproduce the issue with the sample project of this repo?

franksfleung, 2021-02-10:
@franckbour it works. I found there is a new method and the problem solved

closed 2021-02-10 by franksfleung