franckbour/Plugin.NFC #55: HCE compatibility

giv86, 2020-09-22:
Does this plugin, by any chance, can help me with card emulation?

saamerm, 2020-09-22:
@giv86 could you share some links showing how card emulation is done on iOS native?

giv86, 2020-09-22:
> @giv86 could you share some links showing how card emulation is done on iOS native?

For iOS there is no possibility currently. Only for android. And for such there are some examples, however my programming skills are to low to use them...
https://github.com/xamarin/monodroid-samples/tree/master/CardEmulation/CardEmulation

saamerm, 2020-09-22:
@franckbour since its not supported by iOS yet nor in the foreseeable future, do we want to add this functionality to the library?

giv86, 2020-09-23:
> @franckbour since its not supported by iOS yet nor in the foreseeable future, do we want to add this functionality to the library?

But for android this is fully supported. Not only iOS is on the plate with Xamarin :)

franckbour, 2020-09-23:
> > @franckbour since its not supported by iOS yet nor in the foreseeable future, do we want to add this functionality to the library?
> 
> But for android this is fully supported. Not only iOS is on the plate with Xamarin :)

This is a cross platform library so, in my opinion, it's not interesting to add a feature only available in one platform.
But feel free to make a pull-request to add this feature if you really need it.

saamerm, 2020-09-23:
@giv86 I could have looked into it if it was technologically possible to do in native iOS development (swift/objective-c).
Or you could hire someone to implement it for you, they just have to join this library along with the CardEmulation sample you shared

saamerm, 2020-11-19:
@giv86 should we close this?

closed 2020-12-01 by franckbour