franckbour/Plugin.NFC #59: NFC is not available/disabled on iPhone iOS 14

duggan0f, 2020-10-30:
Hi folks,
I'm running some test in the NFC/RFID domain and found your package.
I managed to build your sample for iOS and loaded on my 2 iPhone (7 and X, both running iOS 14.x).
But I always get the message that the NFC is not available/disabled.
Do you have any advice or solution?
Is it an iOS compatibility reason?

### Basic Information

- Version with issue: 0.1.18 
- IDE: VS for Mac, Xamarin 
- Platform Target Frameworks:
  - iOS: 14.0.1
- Affected Devices: iPhone 7 and X

saamerm, 2020-11-19:
Are you sure you have the right certificate and provisioning profiles that include the NFC entitlement? I obviously needed to change the bundle ID to the my own app who's details are registered in the Apple Developer portal. There is a work around I believe, that doesn't require a paid Developer account using Xcode to generate a temp cert & provisioning profile.

Just for you, I just downloaded the latest sample and built it on my iOS 14 iPhone 8 using the Default SDK 14.2 on Visual Studio for Mac, and it worked. You can see it in the picture:
[image]

Please give me more details, and maybe a detailed message on where you see the error message.

saamerm, 2020-11-19:
@duggan0f ^

Legolas87, 2020-12-04:
i confirm that it doest work for ios 14.2 , is says NFC is not available but actually it is available and works for other nfc apps

sdinnes, 2021-02-09:
I am having the exact same issue, albeit I am using VS for Windows and an iPhone XR.

Can I ask for clarification about "right certificate and provisioning profiles"?  I have a provisioning profile issued for my team and in XCode on the Mac I can verify both that the ID (the ten character label next to my name) shown matches the one used by VS; and that the profile itself includes "Near Field Communication Tag Reading" in Capabilities and "com.apple.developer.nfc.readersession.format" in Entitlements.  My info.plist has all the required entries.

Despite all this, all I ever get is "NFC scanning is not available".  I have an app NFC Taginfo that reads my tag straightaway so I know it is not an issue with the phone.

Finally, I have no issues scanning tags in the Android version of the app.

When you wrote "the right certificate and provisioning profiles" - does this mean there is a certificate required as well as the provisioning profile?  Any help would be really appreciated!

saamerm, 2021-02-09:
Can you guys add breakpoints in the library iOS logic for availability and see what's the reason it's returning false?

franckbour, 2021-02-09:
@sdinnes @Legolas87 Can you provide a reproduction sample?

sdinnes, 2021-02-10:
@saamerm Thanks for the suggestions!  In order to narrow down the issue, I took your sample project from here, gave it the same ID as my app (in order to use and test my provisioning profile) and ran it on my phone.  It worked, perfectly!  So that eliminates the phone and the provisioning profile.

I tried emulating your code in mine, but this made no difference.  CrossNFC.Current.IsAvailable still returns false (although IsWritingTagSupported returns true, oddly).  So I took your Plugin.NFC project code and added it to my app so I could step into it.

This reveals that NFCReaderSession.ReadingAvailable is returning false when called in NFC.iOS.cs.

In the output window in VS, I see this message appear before the calls to the NFC class:

System.InvalidCastException: Unable to cast object of type 'Mono.Debugger.Soft.PointerValue' to type 'Mono.Debugger.Soft.PrimitiveValue'.

I also see this if I inspect the properties of NFCReaderSession, for property class_ptr.  Is this relevant?

sdinnes, 2021-02-11:
I am delighted now to report success!  Having eliminated all code and obvious settings differences between my app and the sample one, I started comparing the likely suspects, the iOS files, and one by one made them the same.  The first significant difference was between the vsproj files: the sample app has this entry missing from mine under PropertyGroup (Debug|iPhone):

< CodesignEntitlements >Entitlements.plist</ CodesignEntitlements >

However, when I put that in, it stopped my app from deploying! ("ApplicationVerificationFailed: Failed to verify code signature of /.....iOS.app : 0xe8008016 (The executable was signed with invalid entitlements.)")

Deciding to plough on, I found a difference in the Entitlements.plist, where mine had an entry for "keychain-access-groups" containing the bundle ID of the app.  Removing this was the final step that fixed the problem and I now have NFC scanning!

Many thanks for the support!

franckbour, 2021-02-11:
@sdinnes glad to hear it!

saamerm, 2021-04-13:
@sdinnes would you like to close this issue?

sdinnes, 2021-04-13:
@saamerm - yes please, you may close this issue now.  Thank you for your help!

closed 2021-04-19 by franckbour