franckbour/Plugin.NFC #60: Events raised for errors

thedee, 2020-11-09:
Is there any event that is raised when a scan error for iOS occurs?  For example if you get a "Stack Error" or a "Read error. Please try again" it shows the error on the iOS popup but I don't see any events that tells me there was an error.

saamerm, 2020-11-19:
@thedee Not yet. Where would you recommend that event getting invoked? Also is there a similar thing we can do for Android as well?

aakybreaky, 2022-11-09:
Hello. Is there something done for this problem? OniOSReadingSessionCancelled and OnMessageReceived are not fired in this case.

thatkidyousaw1, 2023-07-18:
@saamerm  Maybe the plugin can throw an exception under the "CrossNFC.Current.PublishMessage" method for scenarios where these errors are raised? So that atleast appside can handle this gracefully.

thatkidyousaw1, 2023-08-24:
@saamerm Hi! I tried debugging the code, there's no error returned from this code snipped when I am trying to write to an NFC tag that is read-only. It will display a Stack Error dialog though from iOS itself.

[image]

This is from NFC.iOS.cs file.

Do you happen to know where should I put the error checking? Because it will always return null even if Stack Error occured.

saamerm, 2023-08-25:
That's a really good question @thatkidyousaw1 ! You will have to create a PR to update the library with that change most likely 

@franckbour do you have any advice?