franckbour/Plugin.NFC #62: The "Identifier" property in the ITagInfo object is Reversed for iOS

JerVaughan, 2020-11-26:
### Description

When scanning an NFC tag, the "Identifier" property in the ITagInfo object is reversed for iOS, and subsequently shows a different SerialNumber property for the same tag.

The "NFC Tools" app on both Android and iOS show the same serial number when scanning a tag.  Android using Plugin.NFC also matches this serial number.  It's only when I use Plugin.NFC on iOS that I'm seeing the reversal.

### Steps to Reproduce

1. Set a breakpoint on the "Current_OnMessageReceived" method.
2. Use "Quick Watch" in Visual Studio to analyze the ITagInfo parameter object.
3. Compare values between iOS and Android.

### Expected Behavior

I expect to receive the same serial number on iOS and Android and when scanning a tag.  I use the serial number for a database lookup request, so they need to be the same across platforms.

### Basic Information

- Version with issue: v0.1.18
- Platform Target Frameworks:
  - iOS:  14.1
  - Android: 10
- Affected Devices: iPhone SE (2nd Generation) and Samsung Galaxy S9

### Screenshots

**Android**
[image]

**iOS**
[image]

saamerm, 2020-11-26:
@JerVaughan any recommendations on what can be changed in order to fix this?

JerVaughan, 2020-11-26:
> @JerVaughan any recommendations on what can be changed in order to fix this?

For now, I've modified my "Current_OnMessageReceived" method in my app with the following when getting the tag's serial number... I'm just reversing the array if it's iOS.  I've successfully modified your code earlier to add-in the pending fix for reading ISO15693 tags (Issue #41 and Issue #61) until it's released, so I'm fairly comfortable with it.  The "GetTagIdentifier(INFCNdefTag tag)" method in NFC.iOS.cs probably needs to be changed to reverse the byte array before returning the identifier, but I only have ISO15693 tags to test with.

`byte[] identifier = tagInfo.Identifier;			
string serialNumber = Device.RuntimePlatform == Device.iOS ? NFCUtils.ByteArrayToHexString(identifier.Reverse().ToArray()) : NFCUtils.ByteArrayToHexString(identifier);`

saamerm, 2020-11-26:
@JerVaughan
Thanks for that, it is very strange. Is there any documentation that shows that iOS reverses the byte array? I certainly don't think anything in our code causes the reverse isn't it?

franckbour, 2020-12-01:
The tag identifier seems to be reversed only with ISO15693 tags but unfortunately, I have none of these. I cannot confirm that assertion for the moment.

f-richter, 2020-12-03:
I can confirm that assertion. Sorry for non-english language in the screenshots. So you can see that the serial number (Seriennummer) in iOS read with the Plugin.NFC is reversed. @JerVaughan maybe you can create a pull request?

| NFC Tools Android  |  NFC Tools iOS  | Plugin NFC Android  |  Plugin NFC iOS |
|---|---|---|---|
|  [image] | [image]  |  [image] |  [image] |

saamerm, 2020-12-03:
This seems to be an iOS bug because I haven't seen this assertion anywhere in their documentation. Have you guys seen anything that got you to try to do the reverse in the first place?

franckbour, 2020-12-03:
No, I didn't know that.

commit 16257adf (2020-12-08): Update NFC.iOS.cs

Fix #62
closed 2020-12-08 by franckbour