franckbour/Plugin.NFC #63: Opening app with IntentFilter does not work on Android

mdhom, 2020-11-30:
### Description
I want my app to open when one of my tags is scanned. I can't get it to work on Android.

### Steps to Reproduce

1. I write a tag with the following NFCNdefRecord:
```csharp
var record = new NFCNdefRecord {
    TypeFormat = NFCNdefTypeFormat.WellKnown,
    MimeType = MIME_TYPE,
    Payload = NFCUtils.EncodeToByteArray(_writePayload)
};
```
2. I add the IntenFilter in MainActivity:
```csharp
[Activity(Label = "SoundMem", Icon = "@mipmap/icon", Theme = "@style/MainTheme", MainLauncher = true, ConfigurationChanges = ConfigChanges.ScreenSize | ConfigChanges.Orientation | ConfigChanges.UiMode | ConfigChanges.ScreenLayout | ConfigChanges.SmallestScreenSize )]
[IntentFilter(new[] { NfcAdapter.ActionNdefDiscovered }, Categories = new[] { Intent.CategoryDefault }, DataMimeType = Services.Nfc.NfcService.MIME_TYPE)]
public class MainActivity : global::Xamarin.Forms.Platform.Android.FormsAppCompatActivity
```
3. I start the app via debugger once, close it.
4. I scan the previously written tag.

### Expected Behavior
My app opens.

### Actual Behavior
Some default android nfc service opens, showing me the writePayload.

### Basic Information

- IDE: VS2019 Preview
- Platform Target Frameworks:
  - Android: 10.0 - API 29
- Nuget Packages:
 - Newtonsoft.Json (12.0.3)

saamerm, 2020-11-30:
Nice catch 
Were you able to use the project and fix the issue? Do you  ou u know what might be causing the behaviour?

mdhom, 2020-12-01:
Well no, actually that is a real problem for me and my app i am trying to develop, so i was hoping for help 😅 
Good point though, i will try and see if I have the same problem with the sample app.

mdhom, 2020-12-01:
I got it working, with writing the following record:
```csharp
var record = new NFCNdefRecord {
    TypeFormat = NFCNdefTypeFormat.Mime,
    MimeType = MIME_TYPE,
    Payload = NFCUtils.EncodeToByteArray(_writePayload)
};
```
So i changed `TypeFormat` from `WellKnown` to `Mime`. I do not know if this has any caveats, but for now it works for me. Can anyone confirm that this is a valid way to fix this issue?

franckbour, 2020-12-01:
Indeed, this is the only way if you want to open a specific application with a tag. 
Type format should be `Mime` and `MimeType` should be your application id (e.g. `application/com.companyname.yourapp`).

See [Launch app when a compatible tag is detected on Android](https://github.com/franckbour/Plugin.NFC#launch-app-when-a-compatible-tag-is-detected-on-android) (I agree that section isn't clear about this)

mdhom, 2020-12-01:
ok thanks, maybe we could update the description with a little note. Or maybe it was just my stupidity. I'll let you decide 😄

franckbour, 2020-12-01:
I will update the readme ;)

closed 2020-12-01 by franckbour