franckbour/Plugin.NFC #66: Supporting raw tag (not ndef} on iOS

falvado, 2021-01-16:
### Description

### Steps to Reproduce

1. 
2. 
3. 

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue:
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

saamerm, 2021-01-16:
@falvado what's your question?

JerVaughan, 2021-01-26:
> @falvado what's your question?

I think the issue might be that you can't scan a tag that doesn't have any records written to it in iOS (i.e. the "raw tag" reference in the title of this issue?).  I've been troubleshooting this issue myself with ISO 15693 tags... the goal for me is to just get the tag's serial number.

The following code works on Android when looking at the ITagInfo parameter passed to the Current_OnMessageReceived( ) function, whether the tag has any records written or not, but **it does not work on iOS if the tag has no records**:

byte[] identifier = tagInfo.Identifier;			
string serialNumber = Device.RuntimePlatform == Device.iOS ? NFCUtils.ByteArrayToHexString(identifier.Reverse().ToArray()) : NFCUtils.ByteArrayToHexString(identifier);

JerVaughan, 2021-01-26:
After further review, this may be the same issue as #64.

When will these changes, along with issue #62, be published into a new version?

JerVaughan, 2021-01-26:
> After further review, this may be the same issue as #64.
> 
> When will these changes, along with issue #62, be published into a new version?

Upon even further review... it looks like these updates are available in v0.1.19!

saamerm, 2021-02-09:
@falvado can we close this?

franckbour, 2021-02-09:
Yes, we can.

closed 2021-02-09 by franckbour