franckbour/Plugin.NFC #67: Get ITagInfo of Launch Tag

marcellangmaier, 2021-01-26:
Thank you for the great plugin. Easy to use and works very well.
I made an app which is launched on scanning a tag.
The app gets startet as it should be.
But how can I get the Tag Information (I am interrested in the Serialnumber) for this Launch Tag?

I am using this Plugin in my Visual Studio 2019 solution with version 0.1.19 of your plugin.

saamerm, 2021-01-26:
- You have to first assign an event handler function to the event, like this:
‘CrossNFC.Current.OnTagDiscovered += Current_OnTagDiscovered;’
- Then start listening to the NFC
- Once you tap on an NFC device it will call that event handler function and carry the ITagInfo object with it along with all the data you need

marcellangmaier, 2021-02-04:
Thank you for your reply. You are describing the procedure to read an NFC tag inside an app. Thats not my problem.
My problem is the following:
Once my app was launched by touching an NFC tag, I want to be able to know the SerialNumber of the tag which caused the app to start.

saamerm, 2021-02-04:
Ah I see! @marcellangmaier, according to https://stackoverflow.com/a/60205855, you need to override the OnNewIntent in your main activity and you ll be able to get the info of the object that woke up your app

saamerm, 2021-04-13:
@marcellangmaier, any success, do you want to close this?