franckbour/Plugin.NFC #68: Support for BAC authentication with chip

pjsecker, 2021-02-13:
We would like to be able to use this library to authenticate to items such as passport and ID cards which are protected and require BAC authentication to open the chip.

https://en.wikipedia.org/wiki/Basic_access_control

Is this something that could be possible, either with existing implementation or further enhancement.

saamerm, 2021-02-25:
@pjsecker there are a lot of different types of NFC tags in the industry, that follow different standards.
This library only supports what technically possible to authenticate using an Android & iPhone, so primarily .
I have not read any reports of BAC being technologically supported by native phones, so this library can't make it happen either

saamerm, 2021-04-13:
@pjsecker any updates, or do you want to close this issue/