franckbour/Plugin.NFC #71: Problems when using ZXing Barcode Scanner

ManuLin, 2021-03-05:
### Description

I'm facing a really strange issue and I'm thus curious if anyone has also run into a similar issue. I'm using the plugin in our app and it works perfectly on Samsung devices, however, I'm facing the following issue on a Huawei device. 

Our app is listening for NFC tags on a page where I'm also using a bar code scanner (using the ZXing.Mobile plugin). On Huawei devices the app is only able to detect and thus read NFC tags when the barcode scanner is NOT scanning - which in our case is not really an option. As I don't have a reasonable explanation why this could happen (and as we are only facing this on Huawei and Oppo devices) I just wanted to know about your thoughts on this?

By the way, we are using the most recent version of the plugin.

ManuLin, 2021-03-05:
Update: I just found a Github issue for the ZXing.Net.Mobile plugin which seems to be related. Unfortunately it seems like this is more a hardware / manufacturer related issues that on some devices the NFC chip is disabled while the camera is active...

https://github.com/Redth/ZXing.Net.Mobile/issues/664

saamerm, 2021-04-13:
@ManuLin makes sense. Do you want to close this issue then?

ManuLin, 2021-04-13:
@saamerm Yes, I'll close it.

closed 2021-04-13 by ManuLin