franckbour/Plugin.NFC #72: NFCV Reading Issue

adnan-rolustech, 2021-03-25:
### Description
I am not able to read the NFCV class of tags. Its isSupported bit is always true

[image]

saamerm, 2021-04-13:
@adnan-rolustech is it supposed to be supported by both iOS & Android?
Could you please share some documentation supporting that?
Also did you try to clone the project and use it instead of the Nuget to debug why it's always false?

adnan-rolustech, 2021-04-13:
> @adnan-rolustech is it supposed to be supported by both iOS & Android?
> Could you please share some documentation supporting that?
Well, I don't have any references but I need to implement NFC-V on a cross-platform. 

> Also did you try to clone the project and use it instead of the Nuget to debug why it's always false?
I have tried by cloning the repo. It works fine for a simple NFC but for NFC-V ISO15693 Tags it always gives this error.

saamerm, 2021-04-13:
As you mentioned, NFC-V is basically just `ISO 15693` which is supported by both [iOS](https://developer.apple.com/documentation/corenfc/nfciso15693tag) & [Android](https://developer.android.com/reference/android/nfc/tech/NfcV) natively.

Are you able to use a different NFC reader app that is already available in the App Stores?

If you look at [this pull request](https://github.com/franckbour/Plugin.NFC/commit/8a781c65dc7d93a6f9526eedfd97fb81347d72ae), it seems like the fix was already made and it was tested by people as well.

Are you having this issue on both iOS & Android devices? What is the error message that you get? Do you have a stack trace?

Tagaishi, 2021-08-05:
Hi,
I'm working on the same type of tags (ISO15963) and have the same problem using the plugin v0.1.20 through Nugget and from the sample project.

Unfortunately I cannot try for IOS as I am only developing for Android and this is were it happens for me.

On Android, I have this app that can fully read my tag: https://play.google.com/store/apps/details?id=keum.daniel25.nfcreader1

The info the original report poster gave was simply the values in the ITagInfo parameter given on the event OnMessageReceived.
But as you can see, the plugin determines that the tag is not supported.
It does read it, UID is correct. But it won't get anything else.

I have tried with fresh new tags "unwritten" (the ones I have always have the last block written on for out of factory check) and also not read before. I also tried written tags to make sure that an almost empty memory wasn't the issue.

Tagaishi, 2021-08-05:
Further note, as I continued to dig by myself.

It seems that in the function GetTagInfo() in NFC.android.cs, the tag java object is transformed to ndef with Ndef.Get(tag).

In my case, the ndef object is null.

From what I found, the only reason it could be null is if the technologies supported by the tag don't include Ndef, 
but after listing those technologies, my tag is able to do NdefFormatable and NfcV.

NdefFormatable means that it is capable of Ndef but hasn't been formated to it when first written on and therefor unusable by the Ndef class.

Solutions are:
 - Use tags that are Ndef formated from the get go
 - Format existing tags before using them
 **- Add NfcV support instead of relying only on Ndef depending on available technologies listed by tag.GetTechList()**

saamerm, 2021-09-19:
@Tagaish thanks for the explanation, could you possibly submit a PR based on your findings?

saamerm, 2021-09-19:
@Tagaishi  thanks for the explanation, could you possibly submit a PR based on your findings?