franckbour/Plugin.NFC #74: Sample for iOS

JKNAUG, 2021-05-04:
How does the initialization in the sample project work for iOS, in MainPage.xaml.cs? Judging from the code, the method CrossNFC.Current.StartListening() is only called when not on iOS, so it would never be invoked on an iOS device. Is the init process different there, or how would it ever work if it doesn't even start listening?
I'm asking because if I tried to do the initialization just like in the sample, and nothing happened on iOS. So I was wondering if the sample is wrong there but also when I add the call to StartListening, nothing happens and there's no exception thrown. Android works just fine.

Am I missing something?

saamerm, 2021-05-04:
@JKNAUG the difference between the two platforms is that unlike Android, iOS doesn't support passive listening.
So in the sample we wanted to show that you can be listening for the tag without having the entire UI blocked by the listener.

JKNAUG, 2021-05-04:
Thanks for the clarification, I understand that. I actually tried to call StartListening also on iOS initially, because unlike in the sample I did it just the same for all platforms. But as mentioned above, that didn't help either, nothing happened, no exception. 
I was hoping that the iOS NFC Scanning UI pops up and I suppose that's the expected behavior?

saamerm, 2021-05-04:
@JKNAUG yes. Check your application output for any error messages while debugging and perhaps place an "Exception Catchpoint" to see if there are any suppressed exceptions too.
Also, if you try the attached sample, did you have any success?

JKNAUG, 2021-05-05:
I just tried the sample and it works as expected, so I'm kind of lost. Double-checked entitlements and everything seems to be alright. Stepping through both codes, no exception, no warning, no nothing - one works and the other one just doesn't bring up that UI, no matter what I do. So my only explanation is that the scanning UI interferes with something else I have in my other App.

saamerm, 2021-05-05:
@JKNAUG noice! 
Ok did you remember to add the nuget to your iOS project as well, and did it get added without errors?

JKNAUG, 2021-05-06:
@saamerm I actually made it work now!

Two reasons:
1) The nuget was not added to the platform project; doesn't seem to be a problem on Android, but apparently on iOS.
2) The iOS project did not have the Custom Entitlements property set to entitlements.plist, so I suppose the permission weren't set correctly

Thank you very much for your support!

saamerm, 2021-05-06:
Amazing, you can close the issue if you're done

closed 2021-05-07 by JKNAUG