franckbour/Plugin.NFC #77: Android - not capturing intent if application was in foreground on tag scan

Andrei-NZ, 2021-05-29:
### Description

Library does not seem to capture intent if application is brought to the foreground via IntentFilter (filtering on DataMimeType)

Reason for this seems to be the fact that OnNewIntent is not firing if application is brought in from background.

I am manually applying using the code below, but not sure if this is the correct way to do this and if it is better to move this into library, if this is expected Android behaviour? 

```
            CrossNFC.Init(this);
            if (this.Intent != null)
                CrossNFC.OnNewIntent(this.Intent);
```

### Steps to Reproduce

1. Run sample on Android in debug
2. Move Android app into background - go to home screen. Debugging still running.
3. Scan NFC tag

### Expected Behavior
1. Application is moved to the foreground
2. CrossNFC.Current.OnMessageReceived firing

### Actual Behavior
1. Application is moved to the foreground
2. CrossNFC.Current.OnMessageReceived not firing

### Basic Information

- Version with issue: 0.1.19
- Last known good version: NA
- IDE: VS 2019 MAC
- Platform Target Frameworks:
  - Android: <uses-sdk android:minSdkVersion="21" android:targetSdkVersion="30" />
- Affected Devices: Motorolla G10, latest Android

saamerm, 2021-09-08:
Why are you initializing the 'CrossNFC.Init(this);' again? @Andrei-NZ

saamerm, 2021-09-19:
@Andrei-NZ ^

phamquoctuan1, 2023-07-08:
Hi Mr. @saamerm 
I have the same issue.