franckbour/Plugin.NFC #80: The readonly function does not set the complete nfc tag to readonly

JarJarBinkz, 2021-08-24:
### Description

The CrossNFC.Current.PublishMessage(tagInfo, true) method does not set the entire Nfc tag to readonly. For example, it is possible to add or change records using an NFC Writer on the desktop. And that is exactly what should be prevented. The Plugin.NFC says if you want to rewrite a tag described with CrossNFC.Current.PublishMessage(tagInfo, true) that it is readonly. But with my Desktop USB NFC Writer I can write it because the records are not set to readonly.

Used nfc tag: NXP Mifare Ultralight - NTAG216

### Steps to Reproduce

1. Use an NXP Mifare Ultralight - NTAG (best the NTAG216)
2. Write the Tag with CrossNFC.Current.PublishMessage(tagInfo, true)

### Expected Behavior
The complete NFC tag should be readonly.

### Actual Behavior
The records are not readonly and can be written with a desktop usb nfc writer. It seems that only a kind of flag was set for the write protection that does not protect the records.

### Basic Information

- Version with issue: Plugin.NFC 0.1.20
- Last known good version: -
- IDE: Visual Studio 16.11.1
- Platform Target Frameworks:
  - iOS:  13.4.1
- Nuget Packages: -
- Affected Devices: iPhone Xs Max

### Screenshots

The NTAG216 tag written with the Plugin.NFC and set to readonly. Here you can see that the blocks are not set to readonly.
[image]

Here is a NTAG213 that has already been purchased on the internet as readonly and is completely on readonly.
[image]

### Reproduction Link

saamerm, 2021-09-08:
@JarJarBinkz this is a good catch. Are you able to replicate this with android as well using the cloned code?

saamerm, 2021-09-19:
@JarJarBinkz

JarJarBinkz, 2021-10-06:
@saamerm Sorry for the late answer. Unfortunately, I don't have an Android device to test at the moment. If I get an Android device in the next few weeks, I can test this.