franckbour/Plugin.NFC #81: I am developer at Sperosens, so my problem is that I am unable to make use of the NFC tag reader/scanner. I looked at an example on Github that uses Xamarin Forms franckbour/Plugin.NFC sample and cloned the repository and after that when I try building and running the project the error is "NFC IS DISABLED". I am using iPhone xe mini v14.7.1 and I am running this on Visual Studio on Windows 10.

tbk21, 2021-09-07:
### Description

### Steps to Reproduce

1. 
2. 
3. 

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue:
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

saamerm, 2021-09-08:
@tbk21 most likely it's because you are building the app without the entitlements selected. Check your debug build settings and compare them to the release to see if entitlements is being used in debug

tbk21, 2021-09-08:
[image]

I cloned the repository and this is how the iOS Build looks.

tbk21, 2021-09-08:
Okay, it works now. I also had to change my settings on the "iOS Bundle Signing" from  Automatic Provisioning to Manual  Provisioning.