franckbour/Plugin.NFC #90: Xamarin Forms Android 12 sdk 31 FLAG_IMMUTABLE issue

TheSundayDev, 2022-01-02:
### Description
issue on android 12 when call CrossNFC.Current.StartListening();

### Steps to Reproduce

1. Set target android sdk 31
2. run the application
3. call CrossNFC.Current.StartListening();

[image]

thanks

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue: 0.1.20
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - Android:  12
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

TheSundayDev, 2022-01-14:
hi all, any news?
thanks a lot!!

closed 2022-02-01 by franckbour

franckbour, 2022-02-01:
The new version (0.1.21) is available in CI Feed (MyGet).
Let me know if it fixes your problem.

TheSundayDev, 2022-02-03:
hi, are you going to put also on NuGet?
thanks

commit e78872c7 (2022-02-01): Merge pull request #91 from franckbour/gh-90

Fix #90: Added support for Android 12
commit f970f704 (2022-02-01): Fix #90: Added support for Android 12