franckbour/Plugin.NFC #94: This is not and issue, is a question: Card Emulation

JORGEGO, 2022-02-21:
Can be possible use this excellent plugin as card emulation to send NDEF / NFC messages?

Thanks

santana-jorge, 2022-03-07:
Hi, I'm a newbie. I'm studying NFC in Xamarin and I would like to know if the Plugin.NFC library can be used to send a message from a mobile device to a reader.

I don't know if what I am asking is related to the question being asked.

The purpose is to simulate that the mobile device is a card and that the reader reads a code. 

Is this possible?

Thanks.

saamerm, 2022-03-07:
@JORGEGO @santana-jorge 
The ability to emulate a card does not exist in iOS, and since this is a cross platform library, we focus on functionality that is available on both iOS & Android as much as possible.
You are however, welcome to submit a Pull Request that includes functionality to emulate cards using Android phones