franckbour/Plugin.NFC #95: Add write support for Mifare Classic tags on apple devices in legacy mode

Tofalvy, 2022-02-25:
### Description
NFCNdefReader session supports NFC tag writing based on the 13th slide of this apple presentation:
https://devstreaming-cdn.apple.com/videos/wwdc/2019/715qg6xyd3qgplhd3m/715/715_core_nfc_enhancements.pdf?dl=1

Currently NFCPlugin doesn't support NFC writing on apple devices in legacy mode, IsWritingTagSupported property of NFCImplementation_Before_iOS13 class is always false, and the necessary implementations are missing from that class.

### Steps to Reproduce
It's not possible to write data to Mifare Classic 1k NFC tags with the plugin on apple devices neither in legacy nor in non legacy mode.

### Expected Behavior
NFCPlugin should support writing data to Mifare Classic tags in legacy mode on iphones.

### Actual Behavior
Currently the plugin doesn't support NFCNdefReaderSession based writing.

### Basic Information

- Version with issue: 0.1.21
- Last known good version: unknown
- IDE: Visual Studio 2022
- Platform Target Frameworks:
  - iOS:  15.3.1
  - Android: n/a
  - UWP:  n/a
- Nuget Packages:
- Affected Devices: iphone 7

Tofalvy, 2022-03-03:
Missing implementation is added, PR is submitted, please, review it.

franckbour, 2022-03-06:
PR added to vnext.
Thanks :)

closed 2022-03-06 by franckbour

Tofalvy, 2022-03-06:
Thanks.
When do you plan to release a new version with these changes included in it?

commit 55763f3d (2022-03-06): Some fixes

Closes #95, Closes #89