franckbour/Plugin.NFC #97: Not an issue but a question - Maui support

bgraham1962, 2022-03-10:
Hi,

I was wondering if you plan to produce a Maui version (.net 6.0?) of your plugin?

Thanks in advance
Brian

saamerm, 2022-03-10:
We all really want to. I haven't been able to find a good tutorial to see what's needed and I haven't been able to set up Maui completely in order to test and debug. Do you have any resources or do you want to give it a try and submit a pull request?

bgraham1962, 2022-03-10:
Hi @saamerm,

I would love to help but I have not got a clue how to start. I dont know how to write Xamarin plugins let alone Maui ones.  I read elsewhere that if a Xamarin Plugin does not have any UI then it should be as simple as recompiling in .net6.0 but thats all I have heard.

Regards
Brian