franckbour/Plugin.NFC #99: Android  scanning stops after 30 minutes

marcoseraphin, 2022-03-14:
### Description

The scanning process stops exactly after 30 minutes

### Steps to Reproduce

Create an Android Things app that starts ranging

### Expected Behavior

Scanning continues indefinitely once ranging starts

### Actual Behavior

Scanning stops after 30 minutes

### Basic Information

Mobile device model and OS version

Android 11

Android Beacon Library version

0.1.20

Seems to be similar to this issue with another library:
https://github.com/AltBeacon/android-beacon-library/issues/526 

### Screenshots

### Reproduction Link

closed 2022-03-14 by marcoseraphin

saamerm, 2022-03-14:
@marcoseraphin why did you close it?

marcoseraphin, 2022-03-14:
it was a problem with a beacon library, NOT with the NFC library. I selected the wrong lib. Sorry....