franckbour/Plugin.NFC #100: Xamarin forms NFC.Plugin problem reading tag

IvesBertoli, 2022-03-24:
### Description
I developing app using Xamarin Forms and need read NFC card to identify peoples and machines.
So, starting ready the app not return data, return empty data and flag IsSupported is false.

### Steps to Reproduce
1. Start read tag
2. Return message empty (print)

### Expected Behavior
Return data writed

- Version with issue: 0.1.21
- Last known good version:
- IDE: VS Mac
- Platform Target Frameworks:
  - Android: 11
 
- Nuget Packages: Plugin.NFC
- Affected Devices: All
[image]

### Reproduction Link
OBS: Your sample in Git are error too

OBS2: The smartphone working fine becouse was tested read with another app and work!

saamerm, 2022-03-24:
@IvesBertoli what is OBS and OBS2?

IvesBertoli, 2022-03-24:
Extra Information (Observation)

saamerm, 2022-03-24:
@IvesBertoli what kind of tag are you using and did you see the application output?
"IsSupported" is false, so place some breakpoints and share the steps to see why it is false

IvesBertoli, 2022-03-24:
this screenshot is result of another android app (https://play.google.com/store/apps/details?id=com.ssaurel.nfcreader).

[image]

IvesBertoli, 2022-03-25:
@saamerm, can you help?

saamerm, 2022-03-25:
@IvesBertoli I dont have the nfc tags that you are using unfortunately.

As I asked before, when you were debugging: did you see the application output?
"IsSupported" is false, so place some breakpoints and could share what is the reason that it is returning false?

dev-vinicius, 2022-03-28:
I was having the same issue, after format the tag with NDEF format with another app (https://play.google.com/store/apps/details?id=com.manjul.utility.nfc.writer&hl=pt_BR&gl=US) it starts working. Is there a way to add a method for this ?

Example method in Android: https://developer.android.com/reference/android/nfc/tech/NdefFormatable?authuser=2

saamerm, 2022-04-06:
@dev-vinicius I dont have NFC tags of that type in order to test it, did you want to submit a PR with that functionality?

dev-vinicius, 2022-04-07:
https://github.com/franckbour/Plugin.NFC/pull/102