franckbour/Plugin.NFC #101: Additional steps for accessing ePassports

forzaphil, 2022-04-06:
I am using the Nuget Plugin.NFC in a Xamarin.iOS app. I want to read data from EU and British passports. I have added the following to the info.plist

<key>NFCReaderUsageDescription</key>
	<string>NFC tag to read NDEF messages into the application</string>
	<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
	<array>
		<string>A0000002471001</string>
		<string>A0000002472001</string>
	</array>

and have NDEF and TAG set in the Entitlements.

I check

bool isAvailable = CrossNFC.Current.IsAvailable;
bool isEnabled = CrossNFC.Current.IsEnabled;

and call
            
CrossNFC.Current.StartListening();

I can detect a tag in a passport, but I get a "Tag is not supported" message on screen and the only thing in the ITagInfo is an identifier.

Question: Is it possible to use this plugin to access the data in ePassports, and what extra steps would I need to take?

forzaphil, 2022-04-06:
OK. The more I dig the more I see how complicated it is to extract the data from the RFID is passports. I guess that's not considered to be part of this solution?

saamerm, 2022-04-06:
@forzaphil that is correct. There is only a certain number of RFID tags that are supported by the hardware of iPhones and Android phones. Reading data from passport chips is not something that it is capable of doing.