franckbour/Plugin.NFC #107: Null Reference Exception - Causing crashes on iOS

saamerm, 2022-05-09:
### Description

On some NFC tags, if the tag is new and empty, the library will raise an exception, and apps can crash. I wasn't able to replicate the crash because I dont have a tag that I can empty, but when I looked at the stack trace, it pointed me to the `GetRecords` function in the code.
When reading through the function, this seemed like the only function that could be causing that crash : `Payload = record.Payload.ToByteArray()`, so I verified that the payload on those nfc tags was empty. When the users wrote something on their tags, the exception & crash went away.

So most likely the fix is to just do this 
`Payload = record.Payload?.ToByteArray()`

### Steps to Reproduce

1. Perform a scan of an NFC tag thats new/empty on an iPhone running iOS 15.4+ 
2. See app crash

### Expected Behavior

* App should not crash and should be able to successfully perform a read operation

### Actual Behavior

* App crashes

### Basic Information

- Version with issue: Latest 0.1.22
- Last known good version: N/A
- IDE: N/A not personally able to replicate 
- Platform Target Frameworks:
  - iOS:  Multiple iOS 15.4 versions
- Nuget Packages: Latest 0.1.22
- Affected Devices: iPhone X +

### Screenshots
[image]

### Reproduction Link
N/A

zLordZachy, 2022-05-19:
Same problem. And I agree with fix, that could be solution. 
I think it is really critical bug. IPhone is not able to read NFC. 
This error came with IOS 15.4.1

HoGo72, 2022-06-13:
Same for me

franckbour, 2022-06-14:
PR merged.
Thanks @saamerm.

closed 2022-06-14 by franckbour

HoGo72, 2022-06-14:
I've installed 0.1.23
Scanning new, empty tag: still null pointer exception
  at Plugin.NFC.NFCNdefPayloadExtensions.GetRecords (CoreNFC.NFCNdefPayload[] records) [0x00001] in <59d13a6c12aa408eb586088692362977>:0 
  at Plugin.NFC.NFCImplementation+<>c__DisplayClass51_2.<DidDetectTags>b__2 (CoreNFC.NFCNdefMessage message, Foundation.NSError error) [0x000b2] in <59d13a6c12aa408eb586088692362977>:0 
  at ObjCRuntime.Trampolines+SDActionArity2V64.Invoke (System.IntPtr block, System.IntPtr arg1, System.IntPtr arg2) [0x00016] in /Users/builder/azdo/_work/1/s/xamarin-macios/src/build/ios/native/ObjCRuntime/Trampolines.g.cs:24513 
  at (wrapper native-to-managed) ObjCRuntime.Trampolines+SDActionArity2V64.Invoke(intptr,intptr,intptr)
  at (wrapper managed-to-native) UIKit.UIApplication.UIApplicationMain(int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x00013] in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:69 

Scanning tag with payload is ok.

Regards, Holger.

HoGo72, 2022-06-14:
The null pointer exception is now NFC.ios.cs, line 1171
var results = new NFCNdefRecord[records.Length];
because records is null,.

A line like
if ((records == null) || (records.Length == 0)) return null;
just before is working for me.

Regards, Holger Gothan.