franckbour/Plugin.NFC #115: Android - NFC tag doesn't get detected if phone is already in contact with the tag before tag scanning is initiated

sushant-shidore, 2022-07-06:
### Description
My app has a specific requirement where detection of NFC tag should be done only when user has performed some prior actions and is then ready to read data from an NFC tag. Tag detection and automatic opening of the app using intent filters is not something which I can use.
On iPhones, NFC tag gets detected even when the phone is already in contact with the NFC tag and then NfcSession.BeginSession() is invoked.
On Android phones however, the tag doesn't get discovered in this case. User needs to take the phone away from the tag and bring it back in contact for it to get correctly discovered. This is true for all Android phones we've tested except Google Pixel phones. 

On Pixel phones, the API 'enableReaderMode()' works which is seemingly able to fetch the already-detected NFC tag. However on other Android phones, it doesn't work that way. Apparently tag detection happens only once when the phone is brought in contact with the NFC tag and if at that moment the app is not actively looking for NFC tag, then the event is lost and doesn't re-appear unless the phone is taken away and brought back in contact with the tag.

Moreover, the enableReaderMode() API is observed to be not so reliable on non-Pixel phones and causes tag loss errors more frequently. So I had to go back to the old method 'enableForegroundDespatch()' on non-Pixel phones.
 
### Steps to Reproduce

1. Link scanning of NFC tag on press of a button on the app
2. Put Android phone in contact with the NFC tag to be accessed
3. Tap on the button from step 1

### Expected Behavior
If tag is successfully detected, OnTagDiscovered callback or ActionTagDiscovered event callback should get hit

### Actual Behavior
None of the expected callback or event gets detected until phone is taken away and brought back in contact with NFC tag or when button is tapped on first while phone is far from the NFC tag and then it is brought in contact with the tag.

### Basic Information

- Version with issue: 0.1.23
- Last known good version: 0.1.23
- IDE: Visual Studio 2017 for Mac, version 8.10.15
- Platform Target Frameworks:
  - iOS:  15.5.1
  - Android: 11, 12 
  - UWP:  N/A
- Nuget Packages:
- Affected Devices: Android phones from all manufacturers except Google Pixel phones

### Screenshots: N/A

### Reproduction Link N/A

saamerm, 2022-08-20:
@sushant-shidore if you use another published NFC app on android, what is the behavior you see?

sushant-shidore, 2022-09-04:
I tried the [NFC Tools app](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc&hl=en_IN&gl=US) and it gave mixed observations. The 'Read' feature of the app doesn't work if the phone is already in contact with the NFC tag and then the read action is triggered. However, the app has a feature to type and send mifare commands in its Advanced section. This feature works even when phone is already in contact with the chip and then a command is sent to it.