franckbour/Plugin.NFC #118: Missing Entitlement error on MAUI sample app

thatkidyousaw1, 2022-08-17:
### Description

Missing entitlement is observed when trying to scan NFC tags using an iPhone device, Android version works well.

### Steps to Reproduce

1. Run sample MAUI application
2. Press Start Listening
3. Missing Entitlement prompt is displayed.

### Expected Behavior
NFC Card is scanned successfully
### Actual Behavior
Missing Entitlement prompt is displayed
### Basic Information

- Version with issue:
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

franckbour, 2022-08-20:
Which entitlement is missing?
Did you follow the iOS specifics section in the readme?

thatkidyousaw1, 2022-08-22:
Hi @franckbour ,

I added all the required strings to be added in both Info.plist and Entitlements.plist. I added the correct profile in Apple Developer as well with NFC Reading Capabilities but it is still displaying a "Missing Entitlements" prompt after the default iOS NFC Scanning UI.

Any idea what's causing the issue?

saamerm, 2022-08-22:
@thatkidyousaw1 
Based on what you’re saying, Under “signing capabilities”, you don’t have “Entitlements.plait” included for your “Debug”-> “iPhone” configuration. 
if that doesn’t make sense, share screenshots for you iOS project settings

thatkidyousaw1, 2022-08-22:
Hi @saamerm ,
This is the current configuration.

[image]

I'm running a MAUI project btw so the view may be different.

thatkidyousaw1, 2022-08-23:
Just a follow-up, this is the .csproj

[image]

saamerm, 2022-08-23:
Hmm you're right. Are you just testing the MAUI example provided? @thatkidyousaw1 or do you want to share another git repo that I can test with?

thatkidyousaw1, 2022-08-24:
Hi @saamerm ,

Yes we only tested the MAUI example provided. Can you try running it on your end and check if the same issue occurs?

Thanks alot.

franckbour, 2022-08-24:
@thatkidyousaw1 I just tested the MAUI example and it works as expected!

The csproj file of the example looks like this :

```xml
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFrameworks>net6.0-android;net6.0-ios</TargetFrameworks>
    <OutputType>Exe</OutputType>
    <RootNamespace>Plugin.NFC.Maui.Sample</RootNamespace>
    <UseMaui>true</UseMaui>
    <SingleProject>true</SingleProject>
    <ImplicitUsings>enable</ImplicitUsings>

    <!-- Display name -->
    <ApplicationTitle>Plugin.NFC.Maui.Sample</ApplicationTitle>

    <!-- App Identifier -->
    <ApplicationId>com.companyname.plugin.nfc.maui.sample</ApplicationId>
    <ApplicationIdGuid>F69E3448-C788-44B5-BB90-6D51D7204472</ApplicationIdGuid>

    <!-- Versions -->
    <ApplicationDisplayVersion>1.0</ApplicationDisplayVersion>
    <ApplicationVersion>1</ApplicationVersion>

    <SupportedOSPlatformVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'ios'">
      14.2</SupportedOSPlatformVersion>
    <SupportedOSPlatformVersion
      Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'android'">21.0
    </SupportedOSPlatformVersion>
    <CodesignKey>iPhone Developer</CodesignKey>
    <ProvisioningType>automatic</ProvisioningType>
  </PropertyGroup>

  <!-- Apple Developer info, replace by your own -->
  <PropertyGroup Condition="'$(TargetFramework)'=='net6.0-ios'">
    <CodesignKey>Apple Development: Franck Bour (XXXXXXXXX)</CodesignKey>
    <CodesignProvision>VS: com.companyname.plugin.nfc.maui.sample Development</CodesignProvision>
  </PropertyGroup>

  <ItemGroup>
    <!-- App Icon -->
    <MauiIcon Include="Resources\AppIcon\appicon.svg" ForegroundFile="Resources\AppIcon\appiconfg.svg"
      Color="#512BD4" />

    <!-- Splash Screen -->
    <MauiSplashScreen Include="Resources\Splash\splash.svg" Color="#512BD4" BaseSize="128,128" />

    <!-- Images -->
    <MauiImage Include="Resources\Images\*" />
    <MauiImage Update="Resources\Images\dotnet_bot.svg" BaseSize="168,208" />

    <!-- Custom Fonts -->
    <MauiFont Include="Resources\Fonts\*" />

    <!-- Raw Assets (also remove the "Resources\Raw" prefix) -->
    <MauiAsset Include="Resources\Raw\**" LogicalName="%(RecursiveDir)%(Filename)%(Extension)" />
  </ItemGroup>

  <ItemGroup>
    <ProjectReference Include="..\Plugin.NFC\Plugin.NFC.csproj" />
  </ItemGroup>

</Project>
```

thatkidyousaw1, 2022-08-26:
May I ask for the configuration you have made with iOS Bundle Signing? Seems like our Apple Dev profile won't connect. Here's ours.

`<Project Sdk="Microsoft.NET.Sdk">

	<PropertyGroup>
		<TargetFrameworks>net6.0-android;net6.0-ios;net6.0-maccatalyst</TargetFrameworks>
		<TargetFrameworks Condition="$([MSBuild]::IsOSPlatform('windows'))">$(TargetFrameworks);net6.0-windows10.0.19041.0</TargetFrameworks>
		
		
		<OutputType>Exe</OutputType>
		<RootNamespace>NFCSample</RootNamespace>
		<UseMaui>true</UseMaui>
		<SingleProject>true</SingleProject>
		<ImplicitUsings>enable</ImplicitUsings>

    
    
    <ProvisioningType>automatic</ProvisioningType>
    
		
		<ApplicationTitle>NFCSample</ApplicationTitle>

		
		<ApplicationId>com.kyocera.test.mauinfcsample</ApplicationId>
		<ApplicationIdGuid>9C5AA2B8-ADE4-4A87-BA0C-AB7EF146958B</ApplicationIdGuid>

		
		<ApplicationDisplayVersion>1.0</ApplicationDisplayVersion>
		<ApplicationVersion>1</ApplicationVersion>

		<SupportedOSPlatformVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'ios'">11.0</SupportedOSPlatformVersion>
		<SupportedOSPlatformVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'maccatalyst'">14.0</SupportedOSPlatformVersion>
		<SupportedOSPlatformVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'android'">21.0</SupportedOSPlatformVersion>
		<SupportedOSPlatformVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'windows'">10.0.17763.0</SupportedOSPlatformVersion>
		<TargetPlatformMinVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'windows'">10.0.17763.0</TargetPlatformMinVersion>
		<SupportedOSPlatformVersion Condition="$([MSBuild]::GetTargetPlatformIdentifier('$(TargetFramework)')) == 'tizen'">6.5</SupportedOSPlatformVersion>
		<CodesignProvision />
		<CodesignKey>iPhone Developer</CodesignKey>
	</PropertyGroup>

	<PropertyGroup Condition="'$(TargetFramework)'=='net6.0-ios'">
	  <CodesignKey>Apple Development: XXXXXX (XXXXXXXX)</CodesignKey>
	  <CodesignProvision>VS: com.xxxx.test.mauinfcsample Development</CodesignProvision>
    <CodesignEntitlements>Platforms\iOS\Entitlements.plist</CodesignEntitlements>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Debug|net6.0-ios|AnyCPU'">
    <CodesignEntitlements>Platforms\iOS\Entitlements.plist</CodesignEntitlements>
	  <BuildIpa>False</BuildIpa>
	  <Optimize>False</Optimize>
	  <MtouchProfiling>False</MtouchProfiling>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Release|net6.0-ios|AnyCPU'">
    <CodesignEntitlements>Platforms\iOS\Entitlements.plist</CodesignEntitlements>
	  <BuildIpa>False</BuildIpa>
	  <MtouchProfiling>False</MtouchProfiling>
	  <Optimize>True</Optimize>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Debug|net6.0-ios|AnyCPU'">
	  <CodesignEntitlements>Platforms\iOS\Entitlements.plist</CodesignEntitlements>
	  <Optimize>False</Optimize>
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Release|net6.0-ios|AnyCPU'">
	  <CodesignEntitlements>Platforms\iOS\Entitlements.plist</CodesignEntitlements>
	  <Optimize>True</Optimize>
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Debug|net6.0-maccatalyst|AnyCPU'">
	  <Optimize>False</Optimize>
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Release|net6.0-maccatalyst|AnyCPU'">
	  <Optimize>True</Optimize>
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Debug|net6.0-windows10.0.19041.0|AnyCPU'">
	  <Optimize>False</Optimize>
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Release|net6.0-windows10.0.19041.0|AnyCPU'">
	  <Optimize>True</Optimize>
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Debug|net6.0-android|AnyCPU'">
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<PropertyGroup Condition="'$(Configuration)|$(TargetFramework)|$(Platform)'=='Release|net6.0-android|AnyCPU'">
	  <MtouchNoSymbolStrip>False</MtouchNoSymbolStrip>
	</PropertyGroup>

	<ItemGroup>
		
		<MauiIcon Include="Resources\AppIcon\appicon.svg" ForegroundFile="Resources\AppIcon\appiconfg.svg" Color="#512BD4" />

		
		<MauiSplashScreen Include="Resources\Splash\splash.svg" Color="#512BD4" BaseSize="128,128" />

		
		<MauiImage Include="Resources\Images\*" />
		<MauiImage Update="Resources\Images\dotnet_bot.svg" BaseSize="168,208" />

		
		<MauiFont Include="Resources\Fonts\*" />

		
		<MauiAsset Include="Resources\Raw\**" LogicalName="%(RecursiveDir)%(Filename)%(Extension)" />
	</ItemGroup>

	<ItemGroup>
	  <PackageReference Include="Plugin.NFC" Version="0.1.26" />
	</ItemGroup>

</Project>
`

franckbour, 2022-08-26:
This is my iOS Bundle Signing configuration (same as yours but the UI is in french)

[image]

thatkidyousaw1, 2022-09-14:
Do you agree that the problem was from my Apple Developer account / permission with NFC? Since we both have the same configurations.

@franckbour

thatkidyousaw1, 2022-09-14:
Hi @franckbour , May I ask for the process on how did you run the MAUI app on iOS? Did you run this via Debug and on VS2022?

franckbour, 2022-09-14:
@thatkidyousaw1 Yes, I run the sample on VS2022 17.3 in debug mode.
In your apple developer account, do you have "NFC Tag Reading" capability enabled in your profile like below ?
[image]

thatkidyousaw1, 2022-09-14:
Yes, this is added already :(

thatkidyousaw1, 2022-09-14:
Hi again @franckbour ,
This is added already :( I don't know why it is still showing the Missing entitlements error 

[image]

franckbour, 2022-09-14:
@thatkidyousaw1 Which version of VS2022 do you use ?

thatkidyousaw1, 2022-09-14:
@franckbour Same as yours which is 17.3

[image]

Could the Apple Developer License not enable NFC Tag Reading on my end? But I can select it when creating Provisioning Profile anyway

thatkidyousaw1, 2022-09-15:
Tried to clone your current project file, still the same issue :(

Time	Device Name	Type	PID	Tag	Message
Sep 15 15:35:49	OPPO-F5	Error	73766	Xamarin.PreBuilt.iOS(libnfshared.dylib)	00000002 8242c3f0 -[NFCTagReaderSession _connectTag:error:]:686  Error Domain=NFCError Code=2 "Missing required entitlement" UserInfo={NSLocalizedDescription=Missing required entitlement}

thatkidyousaw1, 2022-09-15:
@franckbour Hi, Sample App is now working, I will now be closing this ticket.

Updating the info.plist file to this code snippet below did the trick.

`<key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
<array>
    <string>D2760000850101</string>
</array>
<key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
<array>
    <string>12FC</string>
</array>`

closed 2022-09-15 by thatkidyousaw1