franckbour/Plugin.NFC #130: Android 12 and above version issue

saikumarhts, 2023-02-08:
### Description
When we try to read the NFC logger in Android 12 and above versions when calling this method CrossNFC.Current.StartListening(); it is throwing "Java.Lang.IllegalStateException: Foreground dispatch can only be enabled when your activity is resumed"
we are making the call in ViewModel class
 Device.BeginInvokeOnMainThread(() =>
                    {
                        if (CrossNFC.Current != null && CrossNFC.IsSupported && CrossNFC.Current.IsEnabled && CrossNFC.Current.IsAvailable)
                        {
                            try
                            {
                                    CrossNFC.Current.StartListening();
                            }
                            catch (Exception ex)
                            {
                                CrossFirebaseCrashlytics.Current.Log($"ScanNFcViewModel.SubscribeEvents : {ex.Message}");
                            }
                        }
                    });
### Steps to Reproduce

1. Uninstall the app and relaunch it in debug mode
2. in the view model, the above method will be triggered (we are checking some conditions like isAvailable and isSupported then only we are calling this method.   
3. The Exception reporting on Firebase 

### Expected Behavior

### Actual Behavior

### Basic Information

- Version with issue:
- Last known good version:
- IDE: Visual Studio 2022
- Platform Target Frameworks:
 Android 12 and above
- Nuget Packages:
#0.1.26
- Affected Devices:
Oneplus, samsung, moto g
### Screenshots
### Reproduction Link

DunbarChas, 2024-02-17:
I encountered a similar issue, however, mine only started occurring after I tried to write to a card. Did you find a workaround for this by any chance?

andrew-watson-rooster, 2024-02-23:
@DunbarChas I am currently encountering a similar issue when trying to publish a message. I can read a tag just fine with StartListening(), but when I call StartPublishing() and then scan a tag, the builtin Android NFC tag display pops up showing the tag records, and I see the above "Java.Lang.IllegalStateException" in the console.