franckbour/Plugin.NFC #140: While using Navigation in MAUI on Android NFC tag is not scanned

francescobuffetti, 2023-05-26:
### Description
I am making a MAUI application which use Navigation. I have a button in the MainPage which navigates to ScanNFC page. In this page, the scan is not working. If ScanNFC page is the root page, then everything is fine and the tag is scanned.

### Steps to Reproduce

1.  Create two pages. 
2.  Create a button in the first page which pushes to the ScanNFC page
3.  Try to scan the NFC tag

### Expected Behavior
The tag should be scanned as normal. 

### Actual Behavior
The tag is not scanned if i am using navigation, but it is scanned if ScanNFC page is the root page.

Rigo09, 2023-07-20:
Hello, did you manage to solve this error?

Rigo09, 2023-07-28:
Hi @francescobuffetti , I ran into this same problem in a xamarin forms app.  I did was that in the button that opens the page ReadPage.xaml (scanner) I changed:
` private async void OnOpcion1ClickedAsync(object sender, EventArgs e)
{

        await Navigation.PushAsync(new ReadPage());
    }`
by
` private async void OnOpcion1ClickedAsync(object sender, EventArgs e)
{

        await Navigation.PushModalAsync(new ReadPage());
    }`

Maybe this same thing can be giving problems in MAUI

mach-9, 2023-11-02:
Hi Francis,

I'm trying to do the same thing here, but I'm having a problem trying to start the listener on the page load instead of pressing a button to start listening for an NFC tag.

I've tried different measures, in the constructor of the page, but OnResume must be fired after OnCreate. Tried to call it in the constructor, but it fails.

Any luck with this?

Thanks