franckbour/Plugin.NFC #142: Write Tag Android got exception:  Tag IO Error

phamquoctuan1, 2023-07-12:
### Description

### Steps to Reproduce

1. I put breakpoint into MainPage.xaml.cs to debug
2.  then clicked  Write Tag (TEXT) in App 
3.  It was throw exception in ndef.Connect();  in this method WriteOrClearMessage 
                                        went to this line
                                        catch (Java.IO.IOException ioex)
					{
						throw new Exception("Tag IO Error: " + ioex.Message);
					}

### Expected Behavior Can write the tag 

### Actual Behavior  got exception 

### Basic Information Only debug get that, when I run app normally is ok

- Version with issue:
- Last known good version: 
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

saamerm, 2023-09-20:
@phamquoctuan1 Could you please fill out all the information so we can possibly help you fix this issue?

rehnz, 2023-10-31:
I am getting this same error on MAUI with Android ONLY when trying to publish with NFCNdefTypeFormat.Mime TypeFormat.  WellKnown (text) seems to work fine.

[image]

Any guidance?