franckbour/Plugin.NFC #143: IOS / Library not working after published to AppStore Connect / Testflight

hattmann, 2023-09-20:
### Description
IOS / Library not working after published to AppStore Connect / Testflight NFC 

1. Add NFC Plugin to Maui Project
2. In Debug and Local Mode everything runs fine
3. Publish the App To Appstore Connect

### Expected Behavior
NFC Popup raises and NFC Card can be detected

### Actual Behavior
Message "MFC is not available
Followed by Message "NFC is disabled"
Followed by "Session Invalidated / System resource unvailable"

### Basic Information
Content Added to Entitlement.plist:
`	<key>com.apple.developer.nfc.readersession.formats</key>
	<array>
		<string>TAG</string>
	</array>`

Content Added to Info.plist:
`<key>NFCReaderUsageDescription</key>
    <string>Allow access to read MBC NFC Cards</string>
    <key>com.apple.developer.nfc.readersession.felica.systemcodes</key>
     <array>
       <string>8005</string>
       <string>8008</string>
       <string>0003</string>
       <string>fe00</string>
       <string>90b7</string>
       <string>927a</string>
       <string>86a7</string>
     </array>
    <key>com.apple.developer.nfc.readersession.iso7816.select-identifiers</key>
    <array>
        <string>A0000002471001</string>
        <string>A000000003101001</string>
        <string>A000000003101002</string>
        <string>A0000000041010</string>
        <string>A0000000042010</string>
        <string>A0000000044010</string>
        <string>44464D46412E44466172653234313031</string>
        <string>D2760000850100</string>
        <string>D2760000850101</string>
        <string>00000000000000</string>
    </array>`

### Screenshots

### Reproduction Link

hattmann, 2023-09-20:
Ok, it is now working, the "Error" was not a Plugin.NFC issue.

The Line with the Full Path to the `Entitlement.plist` File was wrong, there was no compiler error from .NET Maui for this.
```
<CodesignEntitlements>Entitlement.plist</CodesignEntitlements>
```

The correct Entry with path is now:
```
	<PropertyGroup Condition="$(TargetFramework.Contains('-ios')) and '$(Configuration)' == 'Release'">
		<RuntimeIdentifier>ios-arm64</RuntimeIdentifier>
		<CodesignEntitlements>**Platforms\iOS\Entitlement.plist**</CodesignEntitlements>
		<CodesignKey>iPhone Distribution: Example (GGGNYXS4N)</CodesignKey>
		<CodesignProvision>CompanyAppstoreSigning</CodesignProvision>
		<BuildIpa>true</BuildIpa>
		<IpaIncludeArtwork>true</IpaIncludeArtwork>
		<CreatePackage>false</CreatePackage>
		<IpaPackageName>de.example.hdl</IpaPackageName>
		<UseInterpreter>True</UseInterpreter>
	</PropertyGroup>
````

Sorry for the wrong issue, but perhaps others find this useful.

closed 2023-09-20 by hattmann

saamerm, 2023-09-20:
@hattmann do you want to create a PR to the README.md file so others don't have the issue?