franckbour/Plugin.NFC #147: Using old ISO 15693 non NDEF cards

bgraham1962, 2023-10-24:
### Description
Hi, we are using very old ISO 15693 cards and we are only interested in reading the card ID.  We do not use any of the data sections of the card.  We are using your plugin on Android and iOS.  

On Android it works correctly and returns the ID of the card without error.

On iOS, after a successful scan, we get a green tick but an error message saying "Tag is not supported".  The ID is returned to the app correctly and the app processes it.

We need to remove the error message as the scan was successful. 

Are there any flags to supress this type of message on a successful scan?

### Steps to Reproduce

1. Scan an ISO 15693 non NDEF card.
2. 
3. 

### Expected Behavior

The ID is returned and a Green tick is displayed without any error message.

### Actual Behavior

The ID is returned and a green tick is displayed but an error message "Tag is not supported" is displayed.

### Basic Information
Testing on iOS 16 on iPhone XR
XCode 14.2

### Screenshots
[image]

### Reproduction Link

bgraham1962, 2023-10-27:
I have found the configuration where the error messages can be overridden so I will close this issue.

closed 2023-10-27 by bgraham1962

joefirstcircle, 2025-01-24:
> I have found the configuration where the error messages can be overridden so I will close this issue.

I'm currently facing the same issue with this type of tag - were you able to suppress the `Tag is not supported` error from appearing?