franckbour/Plugin.NFC #151: Apple App Store submission fails validation

donmesserli, 2024-06-05:
### Description

Have an app that is working on a device with a debug build. Created a release build IPA for distribution and when I try to submit it to the App Store, I receive the followig message...

`Asset validation failed (90778)
Invalid entitlement for core nfc framework. The sdk version '17.5' and min OS version '13.0' are not compatible for the entitlement 'com.apple.developer.nfc.readersession.formats' because 'NDEF is disallowed'. (ID: f2f20a6b-7c28-4a7f-a2e9-8fe8233a5e81)`

### Steps to Reproduce

1.  Add Plugin.NFC
2. Add required lines to info.plist and entitlements.plist
3. Make sure scanning works
4. Build release to distribution
5. Submit to Apple App Store

### Expected Behavior
No erros

### Actual Behavior

`Asset validation failed (90778)
Invalid entitlement for core nfc framework. The sdk version '17.5' and min OS version '13.0' are not compatible for the entitlement 'com.apple.developer.nfc.readersession.formats' because 'NDEF is disallowed'. (ID: f2f20a6b-7c28-4a7f-a2e9-8fe8233a5e81)`
### Basic Information

- Version with issue: 0.1.26
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  17.5
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

donmesserli, 2024-06-05:
Searched online and people said to remove the entries from entitlements.plist. This allowed the app to be submitted without error; but IsNFCAvailable returns false.

donmesserli, 2024-06-12:
Checked Apple's documentation. NDEF isn't a valid entry. Just include TAG.

Olaznog15, 2024-07-18:
I'm having similar problems. The nugget works great on Android devices but on IOS is giving me always CrossNFC.Current.IsAvailable == false. I followed all the instructions on the ReadMe, also I tryed to delete NDEFF as suggested above. Nothing works...

The problem is happening on IOS 17.5.1 running on Iphone 11 and on IOS 16.7.8 running on iphone 8 plus

Any ida of what could be happening?