franckbour/Plugin.NFC #153: Tag is not NDEF compliant when publishing

RsZoli, 2024-10-23:
### Description

I'd like to "digitize" a simple printer card, when i read it it really barely has anything on it:

[image]

When i try to publish this read data i got the "Tag is not NDEF compliant" exception.
This is my code:

```
var tag = new TagInfo(_readCard.Identifier)
{
    Records =
    [
        new NFCNdefRecord
        {
            TypeFormat = NFCNdefTypeFormat.Mime,
            MimeType = "application/hu.rszoli.nfctest",
            Payload =_readCard.Identifier
        }
    ]
};

try
{
    CrossNFC.Current.PublishMessage(tag);
}
catch (Exception ex)
{
    await Toast.Make(ex.Message, ToastDuration.Short).Show();
}
```

What am i missing? I do not care any data but the Identifier read off the card, but i do not mind if its NDEF formatted, i just do not understand how to do this!

### Steps to Reproduce

1. Follow above...
2. 
3. 

### Expected Behavior
Publish a tag with a single Identifier and nothing else on it

### Actual Behavior
Tag is not NDEF compliant 

### Basic Information

- Version with issue:
- Last known good version:
- IDE:
- Platform Target Frameworks:
  - iOS:  
  - Android:  
  - UWP:   
- Nuget Packages:
- Affected Devices:

### Screenshots

### Reproduction Link

FaithfulDev, 2024-10-29:
I have little experience with this plugin, so this is just a random guess: Doesn't it say `IsWritable = false`? I would assume that means it was locked and can't be written to anymore.

tentom, 2024-10-29:
In addition to what FaithfulDev said it also says IsSupported = false so I would assume that it is not supported?

RsZoli, 2024-10-29:
Thank you for your response! That is the data from the card that i have successfully read! I only need SerialNumber, nothing else, so the rest is irrelevant! I want to present the data to a NFC reader using my phone, instead of the card itself!

Shaiken, 2024-11-05:
have you succeeded?
i just want to exchange messages, android to android,
but click write tag button, nothing happened

RsZoli, 2024-11-05:
@Shaiken Sadly, not yet!

RsZoli, 2024-11-05:
@FaithfulDev No, because as i have mentioned before, "I want to present the data to a NFC reader using my phone, instead of the card itself!"
So i'm not trying to write any tag!

RsZoli, 2024-11-10:
@FaithfulDev Yes, i'm trying to do HCE exactly! I was under the assumption that writing a tag equals with presenting the same data to a reader, so from the reader's POV it should not matter if it reads the data off a passive tag or the phone, but clearly i have been wrong!

closed 2025-05-08 by RsZoli